realsenseai/librealsense #8009: Getting "rs2::error" and "Couldn't resolve requests" error while generating PCD file through L515

milan-r-shah, 2020-12-16:
| Required Info                 |                                                    |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515                    | 
| Firmware Version                   | 01.05.00.00          | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)   | 5.3.0-28-generic     | 
| Platform                                 | PC                           |
| SDK Version                          |  2.0 (v2.38.1)           | 
| Language                              |  C++                         | 
| Segment			        |  others                      | 

### Issue Description:

I have been trying to generate the PCD file through **Intel RealSense LiDAR Camera L515** using `pcl-color` example code. [command: `./rs-pcl-color`]. However, every time, I get this error:
```
terminate called after throwing an instance of 'rs2::error'
  what():  Couldn't resolve requests
Aborted (core dumped)
```
On the other hand, if I launch the SDK then it works perfectly OK.

Sometimes, I also get this error:
```
terminate called after throwing an instance of 'rs2::error'
  what():  No device connected
Aborted (core dumped)
```
In this case, even SDK can't detect the Lidar. I have to first unplug and then plug it again. Then SDK works but I still can't use `pcl-color` example because I get the first error.

Surprisingly, this `pcl-color` example code works fine with **D435i**. So, now, how can I fix this issue & use `pcl-color` example code with **L515**?

Thank you so much in advance!

RealSenseSupport, 2020-12-20:
Hi @milan-r-shah 

Thank you for bringing this to our attention. We were able to reproduce these issues and will investigate further. 

In the meantime I believe the L515 specific problem u are seeing is due to the pcl sample configuring the depth and infra streams to unsupported formats. Try changing the corresponding calls in main() to...

cfg.enable_stream(RS2_STREAM_INFRARED, 1024, 768, RS2_FORMAT_Y8, 30);
cfg.enable_stream(RS2_STREAM_DEPTH, 1024, 768, RS2_FORMAT_Z16, 30);

...and see if that allows the PCD file to be written.

Thanks

RealSenseSupport, 2021-01-11:
In any case...this PR resolves this:

https://github.com/IntelRealSense/librealsense/pull/7878

Thanks

closed 2021-01-11 by RealSenseSupport