realsenseai/librealsense #8010: is it possible to set the iris/F-number of the RGB camera for D455

changchenliang, 2020-12-16:
Hi. I am using Realsense D455 and pyrealsense script in python to get RGB stream. I want to ask that is it possible to change the Iris/F-number using pyrealsense script?  If possible, what is the command for that? 
or is there a detailed command to set the capture parameters (like Iris, exposure et al) in python pyrealsense? 
Thank you.

MartyG-RealSense, 2020-12-16:
Hi @changchenliang   You cannot change the F Number (which is **f/2.0** on the 400 Series cameras).   You can though influence the light entering the camera by applying a physical **optical filter** over the top of the camera lenses though.  Intel have published a white-paper document on the subject of optical filters, different types and their properties.  

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400

You can though configure values such as manual exposure, gain, etc in pyrealsense2 though.  

**Manual Exposure**
https://github.com/IntelRealSense/librealsense/issues/2683#issuecomment-436395308

https://github.com/IntelRealSense/librealsense/issues/4449

**Gain**
https://github.com/IntelRealSense/librealsense/issues/2886#issuecomment-448112587

changchenliang, 2020-12-16:
OK. Thank you.

MartyG-RealSense, 2020-12-23:
Hi @changchenliang   Do you require further assistance with this case, please?   Thanks!

changchenliang, 2020-12-23:
Hi. No further requirement. Please close this case. Thank you.

MartyG-RealSense, 2020-12-23:
Thanks very much @changchenliang

closed 2020-12-23 by MartyG-RealSense