realsenseai/librealsense #8012: How can I get the exposure time for auto exposure?

sho-ji, 2020-12-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Operating System & Version |   {Win10}  | 
| Platform                                 | PC  |
| Language                            |  {python} | 

### Issue Description
I'd like to get the exposure time for auto exposure, but I'm having a problem because I don't know how to do it.
Please tell me how to get it.

MartyG-RealSense, 2020-12-16:
Hi @sho-ji   The exposure value does not update when auto-exposure is enabled.  To get the real exposure value, you can access it from metadata.   The link below provides Python scripting for doing so.

https://github.com/IntelRealSense/librealsense/issues/5596

sho-ji, 2020-12-16:
@MartyG-RealSense 
Thank you!

closed 2020-12-16 by sho-ji