realsenseai/librealsense #8013: I just want to compile t265 driver, how to modify makefile

liu-jiaming, 2020-12-16:
It takes too long to compile the complete driver on raspberry pi. I just want the t265 driver. How to modify makefile.

ev-mp, 2021-01-05:
@liu-jiaming hello,
T265 module support is an integral part of the SDK and must be built along with the library using `-DBUILD_WITH_TM2=true` flag.  Several Cmake/Make improvements that should improve build time:
- Skip building examples (-DBUILD_EXAMPLES=false) it may take about 30% of the compile time
- Make sure you have swap memory file defined (see RPI installation guide for details).
- Use `make -j$(($nproc)-1))` for parallel build.  Don't use 'make -j' as it will most certainly get the board stuck.

RealSenseSupport, 2021-01-25:
Hi @liu-jiaming 

Were you able to try the information that ev-mp outlined?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2026-10-23 by MartyG-RealSense