realsenseai/librealsense #8014: what is the meaning of get_units() in line 52 of cv-helpers.hpp? why dm  must multiply f.get_units()?

qingweihk, 2020-12-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | D435i
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | Ubuntu20.04
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
what is the meaning of get_units() in line 52 of cv-helpers.hpp? why dm  must multiply f.get_units()?

MartyG-RealSense, 2020-12-16:
Hi @qingweihk  It looks as though the librealsense depth frame data has been converted to CV_64F in the lines just before line 52 and then the script is using the **depth unit scale** of the camera that is provided by **get_depth_units()** to convert to a distance value in meters.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/cv-helpers.hpp#L52

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1depth__frame.html#a23ef301d9526ab3420ec6d4237f44e03

qingweihk, 2020-12-17:
> @qingweihk ，您好：好像librealsense深度框架数据已在第52行之前的行中转换为CV_64F，然后脚本使用**get_units（）**提供的摄像头的**深度比例**将其转换为米。****
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/cv-helpers.hpp#L52
> 
> https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1depth__frame.html#a23ef301d9526ab3420ec6d4237f44e03
> 
> 将像素深度值乘以深度单位比例尺是librealsense中常用的以米为单位查找距离的方法。
> 
> [＃2348](https://github.com/IntelRealSense/librealsense/issues/2348)

Thanks a lot. So f.get_unit() should be a fix number right? But I try to print f.get_unit() three times and got different values as following:  -1194975904  ;   529629648  ;  1001550704 . Is there anything wrong?

MartyG-RealSense, 2020-12-17:
The depth scale (which is 0.001 by default) should be fixed unless it is deliberately changed, yes.  Knowing that it is fixed unless set to a different value, it is usually possible to hard-code a numeric multiplication value (for example in this case, **dm * 0.001**) instead of retrieving the depth scale with an instruction such as get_units().

qingweihk, 2020-12-17:
> The depth scale (which is 0.001 by default) should be fixed unless it is deliberately changed, yes. Knowing that it is fixed unless set to a different value, it is usually possible to hard-code a [...]

Thanks a lot. Further more,I wonder  different value from get_units() means what? Is that meaning something wrong in my camera?

MartyG-RealSense, 2020-12-17:
Your camera likely does not have a problem with its scale.  Are you printing the value of just **f.get_units()** or the entire equation of **dm * f.get_units()** please?

qingweihk, 2020-12-17:
> Your camera likely does not have a problem with its scale. Are you printing the value of just **f.get_units()** or the entire equation of **dm * f.get_units()** please?

I just print  f.get_units() three times  ,and got different values every time : -1194975904 ; 529629648 ; 1001550704

MartyG-RealSense, 2020-12-17:
To be honest, I do not know.   f.get_units() has been used so infrequently in the past that there is not enough reference material for me to provide an accurate answer.

qingweihk, 2020-12-17:
> To be honest, I do not know why this is occurring for you. **f.get_units()** has been used so infrequently in the past that there is not enough reference material for me to provide an accurate answer.

thanks

closed 2020-12-30 by MartyG-RealSense