realsenseai/librealsense #8015: some problem about application

bladesaber, 2020-12-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 48.0.4)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.}                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Dear anyone:
I am doing a project about use D435 camera in indoor at night time, because of the demand of customer, there is no light in the environment, so I decide to use IR image. But I find some problems and hoping help:
(1) when I get image from left or right camera, there is a lot of light spots which cause by laser emitter, how can I remove them
(2) I try to set rs.option.laser_power to low value which can remove the light spot, but it seems that it will affect the quality of depth image, so I think may be it is not a good idea, do you have some better ideas?
(3) can I use external IR Illuminator, if I use other external IR Illuminator, does it be help to remove light spot or it will do harm to depth image?

MartyG-RealSense, 2020-12-16:
Hi @bladesaber   Yes, the **Laser Power** value is connected to the depth image.   The lower the Laser Power value, the more sparse in detail the depth image becomes.

It sounds as though the multiple spot effect that you are describing is **laser speckle**, a noise phenomenon on the image that results from the laser-based projector of the 400 Series cameras.

If you are using the camera indoors at night in complete darkness, you will likely need some kind of IR light source, whether from the built-in projector, an external projector or an IR illuminator.  The link below provides advice about IR illuminators from the Chief Technical Officer of the RealSense Group at Intel (**agrunnet**):

https://github.com/IntelRealSense/librealsense/issues/2000

agrunnet confirms in that discussion that an IR Illuminator's light-source can provide light that can be used by the camera to analyze depth in a scene without a dot pattern.  So an illuminator would seem to meet your particular project needs.

bladesaber, 2020-12-18:
thank you for useful response

closed 2020-12-18 by bladesaber