realsenseai/librealsense #8017: Best resolution setting for D435i and L515?

oym1994, 2020-12-16:
* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

|---------------------------------|------------------------------------------- |
| Camera Model                       | { D453i L515 } | 
| Operating System & Version |   {ubuntu18.04  | 
| Kernel Version (Linux Only)    |  (5.4)                                         | 

Hi, what is the best resolution setting for D435i and L515? I just vaguely remember for D535i .it is 848 * 480 but not sure.
Thank you!

RealSenseCustomerSupport, 2020-12-16:
----------------------------------------------

Hi @oym1994 Yes. For D435i, the best resolution is 848x480

Another support engineer will answer the question on L515.

###

RealSenseSupport, 2020-12-16:
Hi @oym1994 

For L515, the best resolution is 640x480

oym1994, 2020-12-18:
Thanks for ur kind response!

oym1994, 2020-12-18:
> Hi @oym1994
> 
> For L515, the best resolution is 640x480
Thank you! But the 640*480 is a invalid resolution for L515 rgb!

RealSenseSupport, 2020-12-21:
Hi @oym1994 

Yes on the L515 the VGA resolution stated was meant to be for the depth not for the RGB.  RGB wise you have 1920x1080, 1280x720 and 960x540 available.

Do you need anything else?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-08 by RealSenseSupport