realsenseai/librealsense #8019: Using Distance Between 3D Points to Calculate Angle to Object in Depth Frame

bmils, 2020-12-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.12.9.0 | 
| Operating System & Version |   Win (10) | 
| Platform                                 | PC |
| SDK Version                            |  2.4.0                         | 
| Language                            |  C#  | 

### Issue Description
I am currently working on a snippet of code that takes two 3D points in the Depth Frame and calculates the distance between them. I am then trying to utilize SSS triangle math and the angle version of the Law of Cosines to calculate the angle towards the object in the frame. I also looked at and modeled somewhat close to the math used in the rs-measure example.

Here is what I have so far:
```C#
int x1 = index * 10; //where index is 0 to 63
//y is not declared here because this snippet is in a loop where y is 0 to depthF.Height - 1
int x2 = depthF.Width / 2; //half width of frame
int y2 = depthF.Height / 2; //half height of frame
//calculate center point depth
double centerPointDepth = Math.Round(depthF.GetDistance(x2, y2), 3);
//get distance between center and newly detected point
double distBetweenPoints = Math.Round(Math.Sqrt(Math.Pow((x2 - x1), 2) + Math.Pow((y2 - y), 2) + Math.Pow((centerPointDepth - objectDist), 2)), 3); // square root of (x2 - x1)^2 + (y2 - y1)^2 + (z2-z1)^2
//use law of cosines to calculate the angle
double angle = 0.0;
if (centerPointDepth > 0 && objectDist > 0)
{
     angle = (Math.Pow(centerPointDepth, 2) + Math.Pow(objectDist, 2) - Math.Pow(distBetweenPoints, 2)) / (2 * centerPointDepth * objectDist);
     //now take inverse cosine of angle to get it in radians
     angle = Math.Acos(angle);
     //now convert radians to degrees
     angle = ConvertRadiansToDegrees(angle);
}
```
Is this possible to do with the Depth frame data? Any suggestions or thoughts on my implementation are appreciated.

MartyG-RealSense, 2020-12-16:
Hi @bmils   The information in the link below about using 3D points to determine angles using a **plane fit** may be helpful to you.

https://support.intelrealsense.com/hc/en-us/community/posts/360050894154/comments/360013322694

bmils, 2020-12-16:
Thanks for the reply @MartyG-RealSense, I appreciate you taking the time to try and help.

[This](https://www.geeksforgeeks.org/find-all-angles-of-a-triangle-in-3d/) was able to help me out for anyone interested or looking to solve a similar problem.

closed 2020-12-16 by bmils

MartyG-RealSense, 2020-12-16:
Thanks very much for sharing the details of what was helpful to you.  Good luck with your project!