realsenseai/librealsense #8020: ERROR [2895107136] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY

mbobinger, 2020-12-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Raspbian latest}  | 
| Kernel Version (Linux Only)    |  (latest)                                         | 
| Platform                                 | RaspberryPi  |
| SDK Version                            |  { intel realsense / 2.36.0 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {others }                          | 

Hey guys, I have a couple of D435 in use along with a RPI4 - 4GB. My latest error that I have already posted here was:
```
UNKNOWN in rs2_create_device(info_list:0x20bd070, index:0):
failed to set power state
```
I thought I have already fixed this by implementing a restart code as follows:
```
    # Initialize the realsense stream
    intel_boolean = False
    for i in range(5):
        if intel_boolean == False:
            try:
                print("trying to init intel realsense")
                pipeline = rs.pipeline()
                config = rs.config()
                intel_boolean = True
                print("intel realcam initialized")
            except:
                print("intel cam init failed")
                ctx = rs.context()
                list = ctx.query_devices()
                for dev in list:
                    # serial = dev.query_sensors()[0].get_info(rs2.camera_info.serial_number)
                    # compare to desired SN
                    dev.hardware_reset()
                    intel_boolean = False
        else:
            break
    if intel_boolean == False:
        print("Checked and reset the camera " + str(i) + " times without success, now rebooting")
        os.system("sudo reboot")
    else:
        print("Camera worked fine after so many iterations: " + str(i))
```

So in any case that fixed the problem for me but now, I cannot even hardware reset the USB cam again though while starting realse-viewer it tells a camera was recognized but because of the failed to set power state error I can not access the camera and get following error:
`ERROR [2895107136] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY`
The camera has been updated via realsense-viewer a day ago.

Thank you!

MartyG-RealSense, 2020-12-16:
Hi @mbobinger   As a starting point in investigating your case, could you tell me please whether your script above is running when you launch the RealSense Viewer?  Does the problem still occur if you launch the Viewer immediately after boot-up of the Pi without having started any other programs?   Thanks!

MartyG-RealSense, 2020-12-23:
Hi @mbobinger  Do you require further assistance with this case, please?   Thanks!

mbobinger, 2020-12-23:
Thank you @MartyG-RealSense as always for your help, Merry-Xmas!

MartyG-RealSense, 2020-12-23:
You are very welcome @mbobinger    :)   Thanks for the update about your success in finding the solution!

closed 2020-12-23 by MartyG-RealSense

hichamhendy, 2022-01-17:
I am facing the same issue here by launching launch file underneath ROS
I git always the following error Setting Dynamic reconfig parameters. 17/01 13:05:14,487 ERROR [140619041208064] (handle-libusb.h:100) failed to claim usb interface: 5, error: RS2_USB_STATUS_BUSY
I am using
Device info: 
    Name                          : 	Intel RealSense L515
    Serial Number                 : 	f1272491
    Firmware Version              : 	01.05.04.01
    Recommended Firmware Version  : 	01.05.08.01
    Physical Port                 : 	1-4-10
    Debug Op Code                 : 	15
    Product Id                    : 	0B64
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	2.1
    Product Line                  : 	L500
    Asic Serial Number            : 	0003cb01029e
    Firmware Update Id            : 	0003cb01029e

Much worse is that It sometimes freezes the whole OS (Ubuntu) and I would need then to hard-restart the labtop!
I am using:
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic

Can anybody help me?

MartyG-RealSense, 2022-01-17:
Hi @hichamhendy  A response to your question has been posted at https://github.com/IntelRealSense/realsense-ros/issues/2222

hichamhendy, 2022-01-17:
This a separate issue and total different. The camera manages somehow to crush my Laptop. I tried anyway the suggested solution to check if the freezing would still and it is still as it is

hichamhendy, 2022-01-17:
Thank you for response. The issues are different and not correlated. The camera L515 manages somehow to freeze/crush my Laptop even though after I modified the launch file properly as it is suggested

MartyG-RealSense, 2022-01-17:
The information about the L515 configuration is correct and will likely resolve the problem with the default configuration being applied instead of your custom configuration.  The freeze-up is likely related to a second, different issue.  

Does stability improve if you unplug the micro-sized end of the USB cable from the base of the cable, turn the connector around the opposite way and re-insert it (USB-C cables are two-way insertion on the micro end).

hichamhendy, 2022-01-17:
The default configuration fixed the issue "Given stream configuration is not supported by the device! " 

However, unplugging, revolving, and re-plugging camera-end cable didn't fix the freeze!!

MartyG-RealSense, 2022-01-17:
Adding a note that we discussed the freeze further at your other case at https://github.com/IntelRealSense/librealsense/issues/10159#issuecomment-1014703291