realsenseai/librealsense #8021: rs-server can not built on Raspberry Pi 4 buster release

kursattuncel, 2020-12-16:
Hello,

rs-server can not be built on Raspberry Pi 4 buster release.

pi@raspberrypi:~ $ nano /home/pi/librealsense_build/librealsense-master/build/third-party/live/liveMedia/include/RTSPServer.hh^C
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

Followed the guide in "libuvc_installation.md"

Change the line in 45:
from 
cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release
to
cmake ../ -DFORCE_LIBUVC=true -DBUILD_NETWORK_DEVICE=ON -DCMAKE_BUILD_TYPE=release 

Following error is thrown and the build is broken:

/home/pi/librealsense_build/librealsense-master/tools/rs-server/RsRTSPServer.cpp: In constructor ‘RsRTSPServer::RsRTSPClientConnection::RsRTSPClientConnection(RsRTSPServer&, int, sockaddr_in)’:
/home/pi/librealsense_build/librealsense-master/tools/rs-server/RsRTSPServer.cpp:68:37: error: no matching function for call to ‘RTSPServer::RTSPClientConnection::RTSPClientConnection(RsRTSPServer&, int&, sockaddr_in&)’
     , m_fOurRsRTSPServer(t_ourServer)
                                     ^
In file included from /home/pi/librealsense_build/librealsense-master/tools/rs-server/RsRTSPServer.cpp:4:
/home/pi/librealsense_build/librealsense-master/build/third-party/live/liveMedia/include/RTSPServer.hh:157:5: note: candidate: ‘RTSPServer::RTSPClientConnection::RTSPClientConnection(RTSPServer&, int, const sockaddr_storage&)’
     RTSPClientConnection(RTSPServer& ourServer, int clientSocket, struct sockaddr_storage const& clientAddr);
     ^~~~~~~~~~~~~~~~~~~~
/home/pi/librealsense_build/librealsense-master/build/third-party/live/liveMedia/include/RTSPServer.hh:157:5: note:   no known conversion for argument 3 from ‘sockaddr_in’ to ‘const sockaddr_storage&’
/home/pi/librealsense_build/librealsense-master/build/third-party/live/liveMedia/include/RTSPServer.hh:135:9: note: candidate: ‘RTSPServer::RTSPClientConnection::RTSPClientConnection(const RTSPServer::RTSPClientConnection&)’
   class RTSPClientConnection: public GenericMediaServer::ClientConnection {
         ^~~~~~~~~~~~~~~~~~~~
/home/pi/librealsense_build/librealsense-master/build/third-party/live/liveMedia/include/RTSPServer.hh:135:9: note:   candidate expects 1 argument, 3 provided
make[2]: *** [tools/rs-server/CMakeFiles/rs-server.dir/build.make:2637: tools/rs-server/CMakeFiles/rs-server.dir/RsRTSPServer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2270: tools/rs-server/CMakeFiles/rs-server.dir/all] Error 2

MartyG-RealSense, 2020-12-16:
Hi @kursattuncel   The discussion in the link below may be a useful reference for you if you have not seen it already.

https://github.com/IntelRealSense/librealsense/issues/7123

kursattuncel, 2020-12-16:
Thank you @MartyG-RealSense 

I did follow the guide there, I suppose rs-server is broken in 2.40 release. It's broken on Mac as well.

cmake ../ -DFORCE_LIBUVC=true -DBUILD_NETWORK_DEVICE=ON -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false 

fails to produce a binary for 2.40 release in mac os x big sur 11.1 with the following error:

[ 65%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/skybox.cpp.o
[ 65%] Building CXX object tools/rs-server/CMakeFiles/rs-server.dir/RsRTSPServer.cpp.o
clang: warning: -latomic: 'linker' input unused [-Wunused-command-line-argument]
/Users/kursattuncel/librealsense-2.40.0/tools/rs-server/RsRTSPServer.cpp:67:7: error: no matching constructor for initialization of
      'RTSPServer::RTSPClientConnection'
    : RTSPClientConnection(t_ourServer, t_clientSocket, t_clientAddr)
      ^                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/kursattuncel/librealsense-2.40.0/build/third-party/live/liveMedia/include/RTSPServer.hh:157:5: note: candidate constructor not viable: no known conversion
      from 'struct sockaddr_in' to 'const struct sockaddr_storage' for 3rd argument
    RTSPClientConnection(RTSPServer& ourServer, int clientSocket, struct sockaddr_storage const& clientAddr);
    ^
/Users/kursattuncel/librealsense-2.40.0/build/third-party/live/liveMedia/include/RTSPServer.hh:135:9: note: candidate constructor (the implicit copy constructor) not
      viable: requires 1 argument, but 3 were provided
  class RTSPClientConnection: public GenericMediaServer::ClientConnection {
        ^
1 error generated.
make[2]: *** [tools/rs-server/CMakeFiles/rs-server.dir/RsRTSPServer.cpp.o] Error 1
make[1]: *** [tools/rs-server/CMakeFiles/rs-server.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

I'll try to compile 2.39 for mac first. Afterwards, I'll take my chances with Raspberry.

MartyG-RealSense, 2020-12-16:
The RealSense user on the case that I linked to earlier installed the Pi SD card image provided by Intel and copied the files from that.

https://github.com/IntelRealSense/librealsense/issues/7123#issuecomment-675739484

kursattuncel, 2020-12-17:
Thank you. I also did that, and I worked (at first). But I accidentally upgraded the camera firmware to 5.12.9.0 and it stopped working, because that version is particularly compiled with librealsense 2.34.1 which does not work with 5.12.9.0

I'll either downgrade to the required fw version compatible with librealsense 2.34.1 or try to run EtherSense for a network connected camera.

MartyG-RealSense, 2020-12-17:
Due to the requirement for the most recent firmwares at the time of writing this to have a certain minimum SDK version, I would recommend installing firmware **5.12.7.100** (the one before the SDK update requirements were introduced).

kursattuncel, 2020-12-17:
Hi @MartyG-RealSense 

I think I figured it out.

Live555 streaming media server does not keep source code history in their public download area.
The working build for rs-server relies on Live555 2020.04.06 version but can only download the "latest" version during compilation. The latest release breaks the compatibility with rs-server current source code (all versions)

Following patch solved my problem.

--- librealsense-2.38.1/third-party/live555/CMakeLists.txt	2020-08-28 02:54:59.000000000 -0700
+++ ../librealsense-2.38.1/third-party/live555/CMakeLists.txt	2020-12-17 12:39:07.113813636 -0800
@@ -1,7 +1,7 @@
 cmake_minimum_required(VERSION 2.8.2)
 project(live-download NONE)
 
-set(live_file "${CMAKE_BINARY_DIR}/third-party/live555/live555-latest.tar.gz")
+set(live_file "${CMAKE_BINARY_DIR}/third-party/live555/live.2020.04.06.tar.gz")
 
 function(remote_source url)
     if (EXISTS ${live_file})
@@ -30,7 +30,7 @@
 endfunction()
 
 if(BUILD_NETWORK_DEVICE)
-    remote_source( "http://www.live555.com/liveMedia/public/live555-latest.tar.gz")
+    remote_source("http://mirror.sobukus.de/files/src/live555/live.2020.04.06.tar.gz")
 endif()

I'm able to compile rs-server without any problem on Raspberry Pi 4.

Please let the maintainers of the rs-server know about the issue.
--

MartyG-RealSense, 2020-12-18:
Thanks so much @kursattuncel for your patch contribution - I have passed it on.

closed 2020-12-18 by kursattuncel

davidcann, 2021-03-18:
Any idea which version of live555 is compatible with librealsense v2.41.0? 

I successfully built v2.40.0 with live.2020.04.06.tar.gz, but v2.41.0 fails.

Is there a way to determine which version librealsense is expecting or do we just need to try every version listed on [the live555 mirror site](http://mirror.sobukus.de/files/src/live555/)?

MartyG-RealSense, 2021-03-19:
Hi @davidcann   In version **2.42.0** of the SDK a month ago, a version freeze was placed on live555, suggesting that the live555 version has been finalized as stable from this SDK version onwards.

[image]

davidcann, 2021-03-22:
Thanks, @MartyG-RealSense.

For anyone who needs to build rs-server on 2.41.0, I found that the live.2020.12.23.tar.gz version of Live555 works.

MartyG-RealSense, 2021-03-22:
Thanks very much @davidcann for your live555 version guidance - it is excellent to hear that it is working for you.