realsenseai/librealsense #8024: Enabling streaming by default when RealSenseViewer opens

jsdratm, 2020-12-17:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |  N/A                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.40.0                          | 
| Language                            |  N/A                          | 
| Segment			|  N/A                          | 

### Issue Description
I am trying to use the Intel RealSense Viewer v2.40.0 for demo purposes and I was wondering if there is a way to make it start streaming the stereo and RGB video when the application starts without forcing the user to click the two red toggle switches.  I have found the configuration file (realsense-config.json) and didn't see any options in there that would help.  I did notice the support for a commands file, but i could not find any documentation around this so I'm not sure if that could be used to start streaming by default.  Any help would be appreciated!

[image]

MartyG-RealSense, 2020-12-17:
Hi @jsdratm   If you are trying to activate the Viewer automatically for the purposes of a stereo and RGB demo then you could instead launch the **rs-capture** example program with your D415.  This will display a depth and RGB stream at launch without having to apply any settings.

You can find a pre-built executable version of this program by right-clicking on the RealSense Viewer launch icon on the windows desktop and selecting the **Open file location** menu option to open the SDK's **Tools** folder.  Then locate in this folder the file **rs-capture.exe** and launch it with a double-leftclick whilst the D415 is plugged in.

[image]

[image]

jsdratm, 2020-12-17:
Hello @MartyG-RealSense , thanks for that feedback, I didn't know that was an option!  That may work if it is possible to disable the GLSL rendering and processing due to constraints on our platform.  Do you know if that is possible in the RealSense Capture Example?  

If not, we may have to stick with the RealSense Viewer.  Going back to the Viewer, do you know what the Commands XML is used for and is it possible to create an XML file with commands that tell the camera to start streaming when the Viewer starts up?

MartyG-RealSense, 2020-12-17:
As far as I know, the **rs-capture** example does not any use of GLSL (a form of graphics acceleration that offloads processing from the CPU onto the GPU).  The example does make use of OpenGL graphics rendering (also referred to as **GLFW**).

Commands.xml looks as though it is related to the user interface elements (managed by **Imgui**) rather than a command-line input interface.

closed 2020-12-30 by MartyG-RealSense