realsenseai/librealsense #8025: Want to get RAW frames from RGB video using Intel RealSense D415

akashmaity, 2020-12-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi, 
I am trying to get the unprocessed RAW16 images from the RGB video frame. Currently, I am using Intel RealSense Viewer to change the format to RAW16. However, among the total frames saved, some of the frames have half of it black and the other half is in RAW format. I believe there might be some data loss. Is there a way to debug this issue?
[image]

JianWang-CMU, 2020-12-17:
I have the same problem on my side. Greatly you post it.
I also use D415. I use the viewer to save the RAW16 data to the disk, and then use the Matlab SDK to read it. Unfortunately, half of the data is black...

MartyG-RealSense, 2020-12-18:
Hi @akashmaity  and @JianWang-CMU   The link below leads to a discussion about converting RGB RAW16 frames to another RGB format using **cv2.cvtColor**.

https://github.com/IntelRealSense/librealsense/issues/7275

akashmaity, 2020-12-21:
Hi @MartyG-RealSense, it is not the conversion to RGB that I am concerned about. Some of the RAW images have a perfect image for one half, and the left side looks completely dark (Please look at the attached image in the post). I am suspecting that there is some kind of data loss, but I am using USB 3.0 and SSD already.

MartyG-RealSense, 2020-12-22:
Hi @akashmaity   I am not certain how you were receiving RAW16 images with the D415 and RealSense Viewer, as on my own Viewer, trying to stream RAW16 results in the message **The requested format RAW16 is not supported for rendering**.

[image]

Nonetheless, if you are aiming to receive an **unrectified** stream and do not mind it being greyscale then I would recommend using **Y16** for RGB instead.   Y16 is used by the camera for calibration purposes.

[image]

[image]

akashmaity, 2020-12-22:
Hi @MartyG-RealSense,
The RAW16 cannot be rendered in the RealSense Viewer, but the data gets stored as a .bag file.
If RAW16 does not work, then Y16 is our last option I guess.

MartyG-RealSense, 2020-12-29:
Hi @akashmaity   Do you require further assistance with this case, please?   Thanks!

akashmaity, 2020-12-30:
Hi @MartyG-RealSense, Yes the issue is solved if I control the camera directly from Matlab. For future reference, this issue is only present if I save a bag file from the GUI and then read frames in Matlab. However, when I use Matlab to directly read frames from the camera, I get raw16 files as expected without any issues.

MartyG-RealSense, 2020-12-30:
Do you need help with saving a bag file and reading the frames in MATLAB, or are you satisifed with your direct control solution, please?

closed 2021-01-06 by MartyG-RealSense