realsenseai/librealsense #8031: Segmentation fault when trying to calculate pointcloud from depth

banuprathap, 2020-12-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   Ubuntu 14.04.5  | 
| Kernel Version (Linux Only)    |  4.4.0-31-generic                                         | 
| Platform                                 | Com Express module with i7-6600U |
| SDK Version                            |  v2.13.0                          | 
| Language                            |  C++                         | 
| Segment			|  Robot                          | 

### Issue Description
I am trying to read a depth image from the device and generate pointcloud from the depth data. I followed the example code to create a test program but I get segfault when I run it.

``` cpp
    rs2::pipeline p;
    rs2::pointcloud pc;
    rs2::config c;
    c.enable_device(serial_number);
    p.start();

    while (true)
    {
        rs2::frameset frames = p.wait_for_frames();
        rs2::depth_frame depth = frames.get_depth_frame();
        points = pc.calculate(depth);
    }
```
GDB shows that segfault is at `librealsense::get_points_sse` but I am not sure why this is happening. Please help.

Full backtrace from GDB: 

```
Starting program: /home/admin/test 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/i386-linux-gnu/libthread_db.so.1".
[New Thread 0xb763cb40 (LWP 14403)]
[New Thread 0xb6e3bb40 (LWP 14404)]
[Thread 0xb6e3bb40 (LWP 14404) exited]
[New Thread 0xb6e3bb40 (LWP 14407)]
[New Thread 0xb663ab40 (LWP 14408)]
[New Thread 0xb5e39b40 (LWP 14409)]
Device Serial number #832412070337
[Thread 0xb663ab40 (LWP 14408) exited]
[Thread 0xb6e3bb40 (LWP 14407) exited]
[Thread 0xb5e39b40 (LWP 14409) exited]
[New Thread 0xb5e39b40 (LWP 14420)]
[New Thread 0xb6e3bb40 (LWP 14421)]
[New Thread 0xb663ab40 (LWP 14422)]
Device Serial number #815412070825
[Thread 0xb6e3bb40 (LWP 14421) exited]
[Thread 0xb5e39b40 (LWP 14420) exited]
[Thread 0xb663ab40 (LWP 14422) exited]
[New Thread 0xb663ab40 (LWP 14423)]
[Thread 0xb663ab40 (LWP 14423) exited]
[New Thread 0xb663ab40 (LWP 14424)]
[Thread 0xb663ab40 (LWP 14424) exited]
[New Thread 0xb663ab40 (LWP 14429)]
[New Thread 0xb5e39b40 (LWP 14430)]
[Thread 0xb5e39b40 (LWP 14430) exited]
[New Thread 0xb5e39b40 (LWP 14431)]
[Thread 0xb5e39b40 (LWP 14431) exited]
[New Thread 0xb5e39b40 (LWP 14432)]
[New Thread 0xb6e3bb40 (LWP 14433)]
[Thread 0xb6e3bb40 (LWP 14433) exited]
[New Thread 0xb6e3bb40 (LWP 14434)]
[New Thread 0xb5638b40 (LWP 14435)]
[New Thread 0xb4e37b40 (LWP 14436)]
[New Thread 0xb43dab40 (LWP 14437)]
[New Thread 0xb397db40 (LWP 14438)]
dist_to_center: 0

Program received signal SIGSEGV, Segmentation fault.
librealsense::get_points_sse (depth=<optimized out>, size=<optimized out>, pre_compute_x=<optimized out>, pre_compute_y=<optimized out>, depth_scale=<optimized out>, 
    points=<optimized out>) at librealsense-2.13.0/src/proc/pointcloud.cpp:215
215	librealsense-2.13.0/src/proc/pointcloud.cpp: No such file or directory.
(gdb) bt full
#0  librealsense::get_points_sse (depth=<optimized out>, size=<optimized out>, pre_compute_x=<optimized out>, pre_compute_y=<optimized out>, depth_scale=<optimized out>, 
    points=<optimized out>) at librealsense-2.13.0/src/proc/pointcloud.cpp:215
        y_z0 = <optimized out>
        z_x1 = <optimized out>
        xyz13 = <optimized out>
        depth0 = <optimized out>
        y_z1 = <optimized out>
        xyz01 = <optimized out>
        d0 = <optimized out>
        x_y0 = <optimized out>
        xyz02 = <optimized out>
        xyz11 = <optimized out>
        d1 = <optimized out>
        z_x0 = <optimized out>
        xyz03 = <optimized out>
        x_y1 = <optimized out>
        xyz12 = <optimized out>
        i = 0
        point = 0xb1fca008
        mapx = 0xb26d4008
        mapy = 0xb25a7008
#1  0xb7d0feac in librealsense::pointcloud::process_depth_frame (this=0x80dee0c, depth=...)
    at librealsense-2.13.0/src/proc/pointcloud.cpp:425
        res = {frame = 0x81088e0}
        pframe = <optimized out>
        points = <optimized out>
        vid_frame = <optimized out>
        tex_ptr = <optimized out>
        mapped_intr = {width = -1073745152, height = -1208040703, ppx = -3.02288681e-05, ppy = 0, fx = 1.40129846e-45, fy = 7.00649232e-45, model = RS2_DISTORTION_NONE, 
          coeffs = {0, 4.2923764e-34, 4.29232864e-34, -1.99959993, -2.97017832e-05}}
        extr = {rotation = {4.27707293e-34, 4.27702518e-34, -1.99960613, -2.49417917e-05, 0, -2.02403608e-05, -1.99960279, 0, 0}, translation = {-2.96980143e-05, 
            -1.99961472, -1.99961424}}
        map_texture = <optimized out>
#2  0xb7d11a6a in operator() (source=<synthetic pointer>, f=..., __closure=0x821f3a8)
    at librealsense-2.13.0/src/proc/pointcloud.cpp:528
        composite = {<rs2::frame> = {frame_ref = 0xb7f92000}, _size = <optimized out>}
        this = 0xb2ed2008
#3  rs2::frame_processor_callback<librealsense::pointcloud::pointcloud()::<lambda(rs2::frame, const rs2::frame_source&)> >::on_frame(rs2_frame *, rs2_source *) (
    this=0x80d23d0, f=0x821f3a4, source=0x80629f8)
    at librealsense-2.13.0/src/../include/librealsense2/hpp/rs_processing.hpp:73
        src = {_source = 0xb7fffaf0}
        frm = {frame_ref = 0x0}
#4  0xb7d15f28 in librealsense::processing_block::invoke (this=0x80dee0c, f=...)
    at librealsense-2.13.0/src/proc/synthetic-stream.cpp:38
        ptr = <optimized out>
        callback = {invocation = 0x80fc250, owner = 0x80fc250}
#5  0xb7cab0e7 in rs2_process_frame (block=0x80eac20, frame=0x821f3a4, error=0xbffff670)
    at librealsense-2.13.0/src/rs.cpp:1604
        __FUNCTION__ = "rs2_process_frame"
#6  0x0804b524 in invoke (f=..., this=<optimized out>)
    at /include/librealsense2/hpp/rs_processing.hpp:103
        ptr = 0x821f3a4
        e = 0x0
#7  calculate (depth=..., this=0xbffff6c0)
    at /include/librealsense2/hpp/rs_processing.hpp:226
        f = <optimized out>
#8  main (argc=1, argv=0xbffff7c4) at src/test.cpp:53
        frames = {<rs2::frame> = {frame_ref = 0x81adad0}, _size = 2}
        depth = {<rs2::video_frame> = {<rs2::frame> = {frame_ref = 0x821f3a4}, <No data fields>}, <No data fields>}
        width = <optimized out>
        height = <optimized out>
        ctx = {_context = std::shared_ptr<rs2_context> (use count 1, weak count 0) = {get() = 0x805bea0}}
        c = {_config = std::shared_ptr<rs2_config> (use count 1, weak count 0) = {get() = 0x80d5070}}
        serial_number = "815412070825"
        p = {_ctx = {_context = std::shared_ptr<rs2_context> (use count 1, weak count 0) = {get() = 0x80d9728}}, 
          _pipeline = std::shared_ptr<rs2_pipeline> (use count 1, weak count 0) = {get() = 0x80cc040}}
        pc = {<rs2::options> = {_vptr.options = 0x8053d68 <vtable for rs2::pointcloud+8>, _options = 0x80eac20}, 
          _block = std::shared_ptr<rs2::processing_block> (use count 1, weak count 0) = {get() = 0x80df160}, _queue = {
            _queue = std::shared_ptr<rs2_frame_queue> (use count 2, weak count 0) = {get() = 0x80ddcb8}, _capacity = 1}}
        points = {<rs2::frame> = {frame_ref = 0x0}, _size = <optimized out>}

```

MartyG-RealSense, 2020-12-19:
Hi @banuprathap  This is a somewhat difficult case to diagnose, due to the mix of very old software components (14.04 Linux, 2.13.0 librealsense, 4.4 kernel) and a board - COM Express - that has no reference information available regarding use with librealsense.

The log would suggest that the problem begins with an error related to SSE, but the  i7-6600U processor on your COM Express board should be modern enough to support SSE (and AVX too).

The starting point that I would recommend for investigating this case is to try a **libuvc backend** build of librealsense to bypass the Linux kernel, so that an age-related conflict with the kernel or the Linux version can be ruled out as a cause of the problem.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

banuprathap, 2020-12-21:
Thanks for your response.

I'm not tied to v2.13.0 and I simply chose it because it was the last version with mention of u1404 support in release notes.
 tried to compile v2.40.0 in the same environment and I ran into a different issue. https://github.com/IntelRealSense/librealsense/issues/8039.

I don't understand your RSUSB suggestion. I was able to apply patches to the kernel module following the instructions in this repo. Are you saying the SSE error can be avoided with bypassing the kernel patching?

MartyG-RealSense, 2020-12-21:
Trying RSUSB is useful to help to eliminate whether a conflict with a particular Linux version or kernel might be responsible for a problem.

I am not familiar enough with SSE to comment on the effect that using RSUSB would have on an SSE related error.

banuprathap, 2020-12-22:
I tried SDK v2.40.0 with RSUSB enabled and disabled. I see the same segfault with SSE. Please help.

```
(gdb) bt
#0  librealsense::pointcloud_sse::depth_to_points (this=<optimized out>, output=..., depth_intrinsics=..., depth_frame=..., depth_scale=<optimized out>)
    at librealsense-2.40.0/src/proc/sse/sse-pointcloud.cpp:96
#1  0xb7bd38ea in librealsense::pointcloud::process_depth_frame (this=0x81a360c, source=..., depth=...)
    at librealsense-2.40.0/src/proc/pointcloud.cpp:257
#2  0xb7bd59c3 in librealsense::pointcloud::process_frame (this=0x81a360c, source=..., f=...)
    at librealsense-2.40.0/src/proc/pointcloud.cpp:382
#3  0xb7bdf4fd in operator() (source=..., f=..., __closure=0x81a8474)
    at librealsense-2.40.0/src/proc/synthetic-stream.cpp:75
#4  rs2::frame_processor_callback<librealsense::generic_processing_block::generic_processing_block(char const*)::<lambda(rs2::frame, const rs2::frame_source&)> >::on_frame(rs2_frame *, rs2_source *) (this=0x81a8470, f=<optimized out>, source=0x81a4590)
    at librealsense-2.40.0/src/../include/librealsense2/hpp/rs_processing.hpp:128
#5  0xb7bdcbc0 in librealsense::processing_block::invoke (this=0x81a360c, f=...)
    at librealsense-2.40.0/src/proc/synthetic-stream.cpp:43
#6  0xb7ce3b94 in rs2_process_frame (block=<optimized out>, frame=<optimized out>, error=0xbffff504)
    at librealsense-2.40.0/src/rs.cpp:2077
#7  0x0804c047 in invoke (f=..., this=0xbffff564)
    at include/librealsense2/hpp/rs_processing.hpp:273
#8  process (frame=..., this=<optimized out>)
    at include/librealsense2/hpp/rs_processing.hpp:354
#9  calculate (depth=..., this=<optimized out>)
    at include/librealsense2/hpp/rs_processing.hpp:433
#10 main (argc=1, argv=0xbffff694) at src/test.cpp:49
```

MartyG-RealSense, 2020-12-22:
Could you try the script in the link below please, which is structured similarly to yours.

https://github.com/IntelRealSense/librealsense/issues/2488#issuecomment-427814215

banuprathap, 2020-12-28:
Thanks for the suggestion. It still fails with segfault when ` pc.calculate(depth)` is called.

Do you have anymore updates on the SSE specific issues I mentioned in the backtrace? Also, I should highlight that we are running the **32bit version** of the OS.

I was able to **disable SSSE3** by commenting out the `-mssse3` flag in [this](https://github.com/IntelRealSense/librealsense/blob/4f37f2ef0874c1716bce223b20e46d00532ffb04/CMake/unix_config.cmake#L21) block.

I no longer face the segfault as code never enters [this](https://github.com/IntelRealSense/librealsense/blob/4f37f2ef0874c1716bce223b20e46d00532ffb04/src/proc/sse/sse-pointcloud.cpp#L63) section.

I checked contents of `/proc/cpuinfo` and ssse3 is listed under supported flags. I am planning to update the OS to 16.04 32bit on the Com Express to rule out hardware support issues. 

In the meantime, do you know the implications of disabling SSSE3 optimization in the SDK? We are planning to connect 4 cameras to the host machine and compute point clouds.

MartyG-RealSense, 2020-12-28:
Thanks very much for the update of the success achieved by disabling SSSE3.  I do not have any further information that I can add regarding your backtrace, as the subject of SSE is mostly outside of my personal knowledge.  

My understanding is that disabling SSE may have implications in regard to the **align processing block**, though not program-breaking ones.  It apparently uses SSE optimization if the computer's processor supports it, otherwise it uses unoptimized align.  If **CUDA** support is enabled in librealsense on devices equipped with an Nvidia GPU then CUDA is used for alignment instead.

closed 2021-01-10 by MartyG-RealSense