realsenseai/librealsense #8032: Wrong timestamp for last recorded frame

ssheorey, 2020-12-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 / L515 | 
| Firmware Version                   | 05.12.08.200 / 01.05.02.00 | 
| Operating System & Version |  Linux (Ubuntu 18.04) / MacOS 10.15.7 | 
| Kernel Version (Linux Only)    |  5.4.0.58                            | 
| Platform                                 | PC  |
| SDK Version                            |  2.40.0   | 
| Language                            |  C++           | 
| Segment			     |                          | 

I'm recording to a bag file with the `rs2::pipeline` API. The last frame gets recorded with a wrong  timestamp and this creates an incorrect duration. While reading the bag file, the reader code (including `realsense-viewer`) waits for ever or crashes. This happens on both platforms listed above (macOS with L515 and Ubuntu 18.04 with D435).

[image]

The debug logs created with `LRS_LOG_LEVEL=DEBUG` are attached.

[Playback_2020-12-18-15_19_17.log.zip](https://github.com/IntelRealSense/librealsense/files/5718767/Playback_2020-12-18-15_19_17.log.zip)
[Recording_2020-12-18-15_11_26.log.zip](https://github.com/IntelRealSense/librealsense/files/5718773/Recording_2020-12-18-15_11_26.log.zip)

ssheorey, 2020-12-22:
This error happens if `pause()` recording is called very soon after `pipeline::start()`.

RealSenseSupport, 2021-01-14:
Hi @ssheorey 

In looking at this further, I think what is happening is the pause function is pausing the recording but not stopping the streams from continuing which could lead to a change in timestamp when recording is resumed.

Took a look at recording via the rs-recorder as well as the recording function inside realsense-viewer and I don't see this issue.

*If we don’t hear from you in 7 days, this issue will be closed.

ssheorey, 2021-01-15:
Thanks for looking into this. My earlier code called `pipeline::start()` immediately followed by `pause()` to wait for the user to request start recording. I wanted to retrieve frames for display while recording was paused. The timestamp issue only affects the last frame recorded, which is strange, and the last timestamp actually looks to be a junk value (e.g. 5 years in the future), not what you would get if the clock had just continued running. This pattern isn't used in `rs-recorder` so the issues doesn't appear. 

After a lot of trial and error, my work around was to create a new pipeline when I needed to start recording, instead of keeping recording paused().

RealSenseSupport, 2021-01-28:
Thanks and glad that you were able to determine a workaround for this.

Is there anything else you need here?

*If we don’t hear from you in 7 days, this issue will be closed.

aangerma, 2021-02-24:
hello @ssheorey 
I tried to reproduce the issue with the flow you described without success , can you please attach a code snippet from your application that reproduce this issue?

closed 2021-03-23 by RealSenseSupport