realsenseai/librealsense #8034: failed to set power state

sbilgi, 2020-12-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (2.34) | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17)   | 
| Kernel Version (Linux Only)    |  (4.15 / 5.9.8)                                         | 
| Platform                                 | aarch64.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {c++}                          | 
| Segment			|  {depth sensing}                          | 

### Issue Description

I try to get live stream (one IR Image and depth map) from D430 with resolution 1280x720. But it gave `Frame didn't arrive within xxxms ` and after when it gave few times consecutively I disable all stream, make hardware reset and reconnect the camera. But this times it gave `failed set power state error`. I have reboot the device this time. Sometimes the error occurs in 1 hour sometimes 7 hour sometimes after a day. Itried with different cameras. All is same.

Is it normal? 
How can I handle this state?
What is the main reason of this error?

MartyG-RealSense, 2020-12-20:
Hi @sbilgi   I note that in the specification at the top of this case, you list **2.34** for the firmware.  Do you mean that you are using RealSense SDK (librealsense) version 2.34, please?

If you are, I recommend updating to a newer version of librealsense if possible, as SDK 2.34 had a problem with continuously generating timing errors.

sbilgi, 2020-12-20:
@MartyG-RealSense yes I mean sdk version. Okay I will update my library version to 2.40. I will give you a feedback.

sbilgi, 2020-12-23:
Hi @MartyG-RealSense error was gone in sdk 2.40 thanks.

closed 2020-12-23 by sbilgi

MartyG-RealSense, 2020-12-23:
Great to hear, @sbilgi  - thanks very much for the update!