realsenseai/librealsense #8035: Standalone deployment and realsense2.dll

ChrisProlls, 2020-12-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 |  
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC  | 
| Language                            |  C# | 

### Issue Description
We really want to avoid copying the SDK `realsense2.dll` because we assume that is for dev-only purpose.
We plan to deploy a C# / WPF application through many Win 10 computer with connected RealSense.
Any ideas about how to deploy at large scale a standalone application without install anytime the SDK ?

MartyG-RealSense, 2020-12-21:
Hi @chris-wojciechowski   If you are using Windows then you could greatly simplify use of realsense2.dll by installing the Windows version of the full SDK on one machine and copying the pre-built version of the **realsense2.dll** file from the SDK folder location below.  You should then only need to paste the DLL file once into your application (as the same application will be installed on all the Windows computers).

**C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64**

[image]

If you do not wish to integrate realsense2.dll into your WPF application, another approach may be to embed an offline web browser window into your C# application and stream the camera data to that embedded browser panel from the PC that it is installed on (i.e no web connection, just local streaming).

https://github.com/IntelRealSense/librealsense/issues/7362#issuecomment-693462166

MartyG-RealSense, 2021-01-03:
Case closed due to no further comments received.

closed 2021-01-03 by MartyG-RealSense

ChrisProlls, 2021-01-05:
Hi @MartyG-RealSense,
Sorry for my late response.

The problem is not specific to the `realsense2.dll`, but on Windows but absolutely need to modify the PATH.
Can't we avoid this modification ?

MartyG-RealSense, 2021-01-05:
I conducted extensive further research on your question of how to avoid including realsense2.dll but was not able to find a solution, unfortunately.