realsenseai/librealsense #8039: Error compiling SDK v2.40.0 in Ubuntu 14.04

banuprathap, 2020-12-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   Ubuntu 14.04.5 (32 bit)  | 
| Kernel Version (Linux Only)    |  4.4.0-31-generic                                         | 
| Platform                                 | Intel PC  |
| SDK Version                            | 2.40.0                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                        | 

### Issue Description
I ran into a linker issue when compiling SDK from the source. Here are the steps I followed.

``` bash
mkdir build && cd build
cmake .. \
	  -DCMAKE_C_COMPILER=gcc-5 \
	  -DCMAKE_CXX_COMPILER=g++-5 \
	  -DCMAKE_BUILD_TYPE=Release \
	  -DBUILD_EXAMPLES=true \
	  -DBUILD_GRAPHICAL_EXAMPLES=false \
	  -DBUILD_PYTHON_BINDINGS=true \
	  -DPYTHON_EXECUTABLE=/usr/bin/python \
	  -DBUILD_EASYLOGGINGPP=false \
	  -DBUILD_WITH_TM2=false \
	  -DFORCE_RSUSB_BACKEND=false
make
```

##### Error from make:
```
Scanning dependencies of target rs-hello-realsense
make[3]: Leaving directory `/tmp/librealsense2/targetfs-root/targetfs-build/librealsense-2.40.0/build'
make[3]: Entering directory `/tmp/librealsense2/targetfs-root/targetfs-build/librealsense-2.40.0/build'
[ 85%] Building CXX object examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o
[ 86%] Linking CXX executable rs-hello-realsense
/usr/bin/ld: rs-hello-realsense: hidden symbol `__cpu_model' in /usr/lib/gcc/i686-linux-gnu/5/libgcc.a(cpuinfo.o) is referenced by DSO
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
make[3]: *** [examples/hello-realsense/rs-hello-realsense] Error 1
```

cmake version: 3.19.1
make version: 3.81
tool chain: gcc 5.5.0-12ubuntu1~14.04

MartyG-RealSense, 2020-12-21:
Hi @banuprathap  May I first ask whether you have followed the recommended setup steps for Ubuntu 14.04 at the top of the section of the source-code building instructions linked to below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk

If you have followed these instructions, what results do you get if you leave out the top two lines of your CMake instruction?

**-DCMAKE_C_COMPILER=gcc-5 \
-DCMAKE_CXX_COMPILER=g++-5 \**

These lines are not in the librealsense build instructions.

banuprathap, 2020-12-21:
Appreciate the quick response on this.

I followed the linked instructions to install gcc5 and noticed that the version installed on my machine is 5.5.0 and the instructions call for 5.0.0

I did a quickscan of `ppa:ubuntu-toolchain-r/test` and it seems like only 5.5.0 is available now. Do you have latest instructions for u14.04?

PS: I came across a similar issue [here](https://stackoverflow.com/questions/38727800/ld-linker-error-cpu-model-hidden-symbol) but not sure where I would add the linker flag to.

MartyG-RealSense, 2020-12-21:
It is likely that the instructions specify gcc 5.0 simply because of when they were originally written and that it is not a strict requirement.  A search of past librealsense cases found that a number of Ubuntu 14 users were using gcc 5.5.

I also saw the case that you linked to during my research for your problem.  This **hidden symbol `__cpu_model'** error occurs in various non-RealSense discussions, but I could not find any other RealSense examples of it.  This makes it difficult to determine what a RealSense-specific fix would be.

If the build is tripping up over the example **rs-hello-realsense** (which is a text-based example with no graphics), do you get different results if you set non-graphical examples to not be built by setting ** -DBUILD_EXAMPLES=false \**

banuprathap, 2020-12-22:
I tried with `-DBUILD_EXAMPLES=false` flag and the SDK compiled without any errors but any subsequent code I link against the SDK fails with the same error as above.

I installed gcc-6 compiler, by adjusting the original linked instructions, and the error no longer persists.  Looks like an issue with the compiler version I was using (v5.5.0)

MartyG-RealSense, 2020-12-22:
Thanks for the update!   Can you confirm that you are now able to successfully link code to the SDK with gcc-6 after building the SDK, please?

closed 2021-01-05 by MartyG-RealSense