realsenseai/librealsense #8042: v2.4.0, master, and development all reliably crash the viewer when trying to turn on both depth and RGB at the same time.

shutej, 2020-12-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel RealSense L515 | 
| Firmware Version                   | 01.05.02.00 | 
| Operating System & Version |   MacOS Catalina | 
| Kernel Version (Linux Only)    |  10.15.7 |
| Platform                                 |  Intel i5 based MacBook Air |
| SDK Version                            |  tag/v2.4.0, master (34fc284d5), and development (b940b84cf)                          | 
| Language                            |  I just want a point cloud, thanks.  | 
| Segment			|  I just want a point cloud, thanks.  | 

### Issue Description

The viewer sigsegv's when turning on both the depth and RGB sensor at the same time; I built with:

cd build; sudo xcode-select --reset; cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false

As is, I have a lovely device that convinces me that it has promise, right up until it crashes trying to gather both pieces of info that I want at the same time... help.

shutej, 2020-12-21:
Interestingly, it seems when I build from tags/v2.4.0 with the above settings and plug the camera in after starting the viewer, I can see all sensor data in 2D mode and capture a .bag file.  When I go into 3D mode, however, it seems to show two overlaying frames that look to be different scales -- then immediately crashes -- suggesting the source of the sigsegv is in the 3D code?

RealSenseSupport, 2021-01-19:
Hi @shutej 

We will have a look at this.

RealSenseSupport, 2021-03-22:
Hi @shutej 

Are you seeing issues with the latest librealsense release?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-03-30 by RealSenseSupport