realsenseai/librealsense #8043: Unable to run rs-server command

yeric1789, 2020-12-21:
Not able to run command: rs-server
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435i  | 
| Firmware Version                   | 5.12.09.00 | 
| Operating System & Version |  Raspbian GNU/Linux 10 (buster), Release version 10                                    | 
| Platform                                 | Raspberry Pi 4 4GB|
| SDK Version                            | 2.40 |
| Language                            |  {python}                         | 

Hi, 
I am attempting to set up a raspberry pi system similar to the one outlined in this file.  [https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras](url). I have already read similar problems described in the following to issues.
[https://github.com/IntelRealSense/librealsense/issues/7123](url)
[https://github.com/IntelRealSense/librealsense/issues/8021](url)

I have downloaded the pre-built image found on [https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras](url)

After downloading this image I placed on my sd card I wasn't able to effectively run the command 
$ rs-server 
I get the result
bash: rs-server: command not found
So I ran the cmds
$ cd librealsense/build/ && cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_NETWORK_DEVICE=ON

Following advice outlined in one of the issues: [https://github.com/IntelRealSense/librealsense/issues/7123](url)
This did not work either. So I attempted one of the fixes suggested by the person in [https://github.com/IntelRealSense/librealsense/issues/8021](url)
This also did not fix my issue and I reversed the edits I made to the CMake file in librealsense/third-party/live555/

I then attempted to rebuild the entire sdk following the directions in [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk](url)

I ran the cmake command with the necessary modifications.

So far this is what I have working on the Raspberry pi
-I am able to run commands realsense-viewer 
-I am able to record data
-I am able to run command rs-enumerate-devices
-I am able to run the command systemctl restart rs-server.service, and I do not get an error of any sort (or at least no error that prints out to the terminal)
However though I am able to run systemctl restart rs-server.service without any error. When I try to then run rs-server I get the same error as I got before.

I have attempted rebooting and shutdown the system multiple times.

I have also tried this set up on a second raspberry pi 4 4GB and got a similar failure.
I am unsure how to proceed from this point on. Or what I have set up incorrectly. Any help you could provide would be much appreciated.
Please let me know if you need any more details.
Thanks.

MartyG-RealSense, 2020-12-22:
Hi @yeric1789   The best way to read the open-source ethernet networking paper is to jump between sections instead of reading it all from beginning to end.  Although I have not performed this project myself, I will offer a sequence of sections that I would suggest for someone who is using the pre-built SD card image.

**Section 1
Section 2**

For the initial setup, the end of Section 2 should be as far as you need to go if you are using the supplied SD card image.  

Make sure that the Pi is connected to the **host computer** (e,g a laptop) by some means, as described in Section **2.5  First Boot**.  After about 30 seconds, the host computer should detect new network named “PI4 USB Device”.

Now use a terminal **on the host computer** to establish a network connection to the connected Pi board.  On Windows, use **Putty** to do so.  Input the command below into the terminal.

**ssh realsense@10.55.0.1**

The initial password and username are **realsense**

Once you are logged into the Pi, proceed to Section 2.6 of the guide and input the commands in this section into the terminal on the host computer to set up a **static shell**

At the end of this section, once you have restarted the system (using sudo reboot) to apply the new configuration, proceed finally to Section 2.7 of the guide to test the networked camera with the **RealSense Viewer** program.

**************************

I suggest skipping Section 3 completely during initial setup if using the supplied SD card image.  This is because this section contains information on building the project from source code (not relevant for an SD card image user) and what could be considered to be further information to come back to once initial setup is succesfully completed.

Sections 4 to 6 could also be considered Further Information to be read after initial setup.

MartyG-RealSense, 2020-12-29:
Hi @yeric1789   Do you require further assistance with this case, please?   Thanks!

yeric1789, 2020-12-29:
No. Thanks for your help. You may close this case.

closed 2020-12-29 by yeric1789