realsenseai/librealsense #8045: Big gyr bias

xiaotaw, 2020-12-22:
sorry for not filling the format, I will make it up as soon as possible.

I noticed a big gyr bias, about 1 deg/s in z-axis, as depicted below.

[image]

RealSenseSupport, 2020-12-22:
@xiaotaw Which product you are talking about?

xiaotaw, 2020-12-24:
> @xiaotaw Which product you are talking about?

L515

ev-mp, 2020-12-28:
@xiaotaw , the measured offset is within the [IMU specifications](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bmi085-ds001.pdf)
[image]

xiaotaw, 2020-12-30:
Got it, thank you! @ev-mp

closed 2020-12-30 by xiaotaw