realsenseai/librealsense #8047: Frame didn't arrive within 15000

dingbaiyin, 2020-12-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430 | 
| Firmware Version                   | 5.12.6| 
| Operating System & Version |  ubuntu | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi3B+|
| SDK Version                            |  2.36.0                          | 
| Language                            |  {C++/opencv }                          | 
| Segment			| others                     | 

### Issue Description
<The camera reports an error " terminate called after throwing an instance of 'rs2::error' what(): Frame didn't arrive within 15000" when calling>
[image]

MartyG-RealSense, 2020-12-22:
Hi @dingbaiyin   There have been a couple of cases in recent months on Jetson boards (which, like Pi, have an Arm processor) where updating both the firmware and the SDK version caused this error to disappear.

https://github.com/IntelRealSense/librealsense/issues/7597#issuecomment-710770436

If the above solution does not solve your problem, it is worth bearing in mind that the Pi 3b only has a USB 2 port and so its USB data transmission rate will be noticably lower than USB 3.  It also has only 1 GB of memory capacity, and once the real memory is consumed then the Pi will start using slower **swapfile** virtual memory.  So using a lower resolution and / or FPS speed may help to avoid the Pi 3b becoming over-burdened.

dingbaiyin, 2020-12-23:
Thank you very much for your reply, I will continue to test, hope the problem can be solved

MartyG-RealSense, 2020-12-30:
Adding note to keep this case open for a further time period.

MartyG-RealSense, 2021-01-12:
Hi @dingbaiyin   Do you have an update that you can provide about this case, please?   Thanks!

dingbaiyin, 2021-01-15:
sorry

closed 2021-01-15 by dingbaiyin

MartyG-RealSense, 2021-01-15:
No problem @dingbaiyin  -thanks for the update!