realsenseai/librealsense #8053: Indexing to a Frame in a Bag File

collinbrake, 2020-12-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 2.40 | 
| Operating System & Version |   Linux, Ubuntu  | 
| Kernel Version (Linux Only)    |   | 
| Platform                                 | PC |
| SDK Version                            |    2.40                      | 
| Language                            |  python | 
| Segment			|  Robot  | 

Hi All, our team is in the process of exporting data from `.bag` files created with the RealSense SDK to `.avi` video files. We are exporting the color frames from the bag file to video for data labeling. After the color data is labeled, we want to be able to map each labeled color frame back to the `.bag` file to get the corresponding depth map or inertial data for each color frame.

Our approach uses the `get_frame_number` method of `pyrealsense.frame`. The frame number of a color frame is stored off to a csv file. When the bag file is indexed, the frame numbers from the csv file don't match the new frame numbers that are assigned. Looking through the csv file, sometimes a frame number is identical for two consecutive frames, and sometimes it jumps by two or three. 

[image]

These abnormalities are random, and may be inconsistent between two exports of the same bag file.

Does anyone know how frame numbers are assigned, or if there is a reliable way to map frames from an exported video back to a bag file?

MartyG-RealSense, 2020-12-22:
Hi @collinbrake  The discussion in the link below should hopefully answer some of your questions about frame number anomalies.  If you still have questions after reading that link, please feel free to return here to ask further questions.  Thanks!

https://github.com/IntelRealSense/librealsense/issues/6130

collinbrake, 2020-12-22:
@MartyG-RealSense Thanks for the useful resources! Closing issue because I no longer am using this approach.

closed 2020-12-22 by collinbrake

MartyG-RealSense, 2020-12-22:
You are very welcome - thanks @collinbrake for the update!