realsenseai/librealsense #8054: Can the depth and color images of different resolutions obtain the corresponding coordinate positions?

nevynwong, 2020-12-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Operating System & Version | Ubuntu 16  | 
| Platform                                 | PC  |
| Language                            |python }                          | 

### Issue Description
The resolution of the depth image is 848*480, and the resolution of the corresponding color image is 1280*720. Can I find the color coordinates of the points in the depth image by "projectoion"?

MartyG-RealSense, 2020-12-23:
Hi @nevynwong   You can find the depth of a specific color pixel using the instruction **rs2_project_color_pixel_to_depth_pixel**

Because you have different resolutions for depth and color, this instruction on its own may not provide an accurate result.  The approach taken by the script of a RealSense Python user in the link below, using alignment and deprojection, may work better for you.

https://github.com/IntelRealSense/librealsense/issues/6749#issuecomment-654185205

nevynwong, 2020-12-23:
Hi @MartyG-RealSense  This seems to be about the coordinate conversion method of the same resolution, is there any coordinate conversion method about different resolutions?

MartyG-RealSense, 2020-12-23:
If alignment is used, like in the Python script in the link, then the align processing block should automatically make adjustments if there is a difference in resolutions.

nevynwong, 2020-12-23:
Can I understand that the alignment still projects all the coordinates in the depth image into the color image? Using the rs2_project_color_pixel_to_depth_pixel function?

MartyG-RealSense, 2020-12-23:
Depth to color alignment maps the depth coordinates to the color coordinates.  If you are using a RealSense camera model where the depth field of view is larger than the color field of view (D435 / D435i) then some of the outer areas of the depth image may get cut off once aligned to the color field of view.   Cameras where the depth and color field of view are the same size are D415 and D455.

My understanding is that you would use **rs2_project_color_pixel_to_depth_pixel** **OR** depth to color alignment, not both techniques.   If both depth and color had the same resolution then rs2_project_color_pixel_to_depth_pixel may be sufficient.   If you have two different resolutions then depth to color alignment with the processing block is likely to work better for you due to the automatic calculations that the align processing block makes.

nevynwong, 2020-12-24:
@MartyG-RealSense  Thanks, I understand. But I would like to know what is the exact operation about Align? Is there any open source code to learn?

MartyG-RealSense, 2020-12-24:
The SDK has an example Python depth to color alignment script called **align-depth2color.py** that you could study.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

MartyG-RealSense, 2020-12-31:
Hi @nevynwong   Do you require further assistance with this case, please?   Thanks!

nevynwong, 2021-01-02:
I have no more questions. Thank you for your help.

closed 2021-01-02 by MartyG-RealSense