realsenseai/librealsense #8055: How can I use webcam(IntelRealsense) as input instead of mp4 file

laifuchicago, 2020-12-23:
To Author:
I am currently using recorded mp4 file as input to do the vehicle detection on Windows 10. However, if I want to use your IntelRealsense camera D430i as the input. Do you have any idea? In Linux we use dev/video0 as input. How about Windows?
I use -i cam but the error occured.
[image]

Jonathan Sung

MartyG-RealSense, 2020-12-23:
Hi @laifuchicago  It looks as though you are trying to access a video stream (RGB), using the RealSense camera as a video webcam instead of using an mp4 file as the video source.  Is that correct, please?

Before we carry on, it is also important to confirm which model of camera you are using.   Your model is stated as **D430i**.  I wonder though if you actually have a **D435i** that is being mis-identified as a D430i (suggesting that the RGB sensor inside the camera is not being detected by the camera's firmware driver).

If you do have a D435i and the RGB sensor was not detected then you would not be able to access the RGB video stream.

laifuchicago, 2020-12-24:
To Author:
My device is D435i , and my computer can detect that by using Intel Realsense Viewer like the figure below, its RGB
[image]

Jonathan Sung

MartyG-RealSense, 2020-12-24:
I found a similar looking case about **object_detection_demo_ssd_async.py** on the Intel OpenVINO forum where it was recommended to pass **cpu_extension.dll** with the -l condition, like you did with **vehicle-detection-adas-binary-0001.xml -l**

https://community.intel.com/t5/Intel-Distribution-of-OpenVINO/python3-object-detection-demo-ssd-async-py-not-working/m-p/1157608/highlight/true#M13608

laifuchicago, 2020-12-25:
To Author:
Actually I can run with mp4, my problem is how to replace the input with realsense camera.

Jonathan Sung

MartyG-RealSense, 2020-12-25:
I ran tests with RealSense cameras in ordinary non-RealSense RGB video programs such as Skype on two different Windows PCs.  On one PC the RealSense RGB stream worked instantly when plugged in without having to change any settings, like an ordinary USB webcam.  On the other PC the camera was detected but refused to show the RGB video.

I ran the tests both with the RealSense SDK (librealsense) installed and with it not installed.   On the PC that it worked on, the results were the same.   So long as the PC is able to show the RGB then it is enough just to have the RealSense drivers showing as present in the Device Manager - the presence of the RealSense SDK is not necessary.

So it may just be that the Windows PC that you are using is one of the ones where the stream does not work immediately when the camera is plugged in, and a different Windows PC may have different results.   There may be nothing that the RealSense support team can do to correct this if it is a Windows related problem.

closed 2021-01-09 by MartyG-RealSense