realsenseai/librealsense #8058: what is the unit of depth measurement for L515 camera?

liuliu66, 2020-12-23:
I try to recover point cloud from rgbd captured from L515 camera but the depth seems not to be measured as meter. So what is the depth unit for this camera?

I use format z16 for depth stream. thanks!

liuliu66, 2020-12-24:
I found that the depth scale of L515 is 0.00025 which is different from 0.001 (maybe 1 meter = 1000 mm) for D435. So the depth image values should be multiply by 0.25 and the depth is correct in real world

RealSenseSupport, 2021-01-20:
Hi @liuliu66 

The depth scale is different between the D400 series products and the L515 camera.  The scale for L515 is 0.00025 as you stated, this is correct.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-27 by RealSenseSupport