realsenseai/librealsense #8062: How to record multi-D455 simultaneously by the viewer?

Oldog, 2020-12-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (05.12.09.00） | 
| Operating System & Version |   {Ubuntu 18.04} | 
| Kernel Version   |  (5.4.0-58-generic) |                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.40 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {VR }                          | 

### Issue Description
[image]

I'm trying to record multi streams with the realsense-viewer. But the Sync button seems to be disabled (as shown in the picture above). Do I need some further settings or the viewer does not support recording multi-D455?

I also find a python script from the previous issue to grap frames of multiple cameras. But I notice that there is no explicitly sync operation. Does the SDK offer a sync function?

MartyG-RealSense, 2020-12-24:
Hi @Oldog  In the link below, Dorodnic the RealSense SDK Manager explains that the greyed-out button sat the top of the RealSense Viewer's options side-panel relates to **software sync**.  

https://github.com/IntelRealSense/librealsense/issues/3084#issuecomment-455227691 

Each camera that is added to the Viewer's options side-panel though has its own Record button, and activating the record button for each individual camera results in a separate bag-file recording for each of the cameras.

[image]

Multiple cameras can have their stream timestamps synced with a master signal through hardware sync but this is not the same as **recording** from multiple cameras.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras

 So a bag-file recording in the Viewer should capture all of the currently active streams from a particular camera, (e.g depth, infrared and color) but not combine the streams of all cameras into a single bag file.

[image]

For capturing streams from multiple cameras on the same computer into a single recording, it may be more practical for you to use ROS.  Its **rosbag record** function can record all currently available stream 'topics' using its **-all** condition.  You can launch each camera independently in a separate ROS 'terminal' window, and the collective topics from all of the camera sources should get captured by the 'all' function of rosbag record.

https://wiki.ros.org/rosbag/Commandline#rosbag_record

More information about configuring a multi-camera ROS setup with multiple terminals can be found in the tutorial in the link below.

https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/

Once a bag has been recorded with rosbag record, you should be able to play it back and view its data in a ROS interface called **RViz**, as described in the link below by Doronhi** the RealSense ROS wrapper developer.

https://github.com/IntelRealSense/realsense-ros/issues/1076#issuecomment-586702026

MartyG-RealSense, 2020-12-24:
If you prefer a Python-based solution and are okay with saving eac camera into a separate bag file, the link below may be a useful reference.

https://github.com/IntelRealSense/librealsense/issues/2882

Oldog, 2020-12-27:
> If you prefer a Python-based solution and are okay with saving each camera into a separate bag file, the link below may be a useful reference. Starting the recording automatically for each camera with the script would make it easier for each bag to have relatively the same timing, which would be harder to achieve with the Viewer if you had to press each camera's Record button to start and stop the individual recordings.
> 
> #2882

Thanks for the help. Now I have 3 D455 and I connect the pin5 and pin9 with a star-configuration. unfortunately I find that they are not synchronized..The framerate of different cameras are not the same. Is the external signal generator neccessary?

I summarize the conclusions here:
1. For HW sync: disable the auto exposure priority in control of RGB cameras. Set the sync mode in control of stereo module. Set to master to mode 1 and others to mode2. 
2. The realsense viewer can not record several streams at the same time. Users need to start / stop recording one by one manually.
3. Software sync just bundle the different streams by selecting closest frame-sets in time of arrival.

MartyG-RealSense, 2020-12-27:
An external signal generator is not needed if you set the **Inter Cam Sync Mode** of one of the D455 to a value of '1'.  This makes it the **Master** camera that generates a sync pulse from pin 5.  

The other two D455 should both have their Inter Cam Sync Mode set to '3' **(Full Slave)** if you want to sync RGB, or both to '2' **(Slave)** if you just want to sync depth.

MartyG-RealSense, 2021-01-03:
Hi @Oldog  Do you require further assistance with this case, please?    Thanks!

Oldog, 2021-01-05:
> Hi @Oldog Do you require further assistance with this case, please? Thanks!

So appreciate for your help~ 

Although the hardware sync still does not work. I'll use an O-scope to check it. 

Currently no more questions and I close this issue.

closed 2021-01-05 by Oldog

MartyG-RealSense, 2021-01-05:
Thanks very much for the update - good luck with your further tests!