realsenseai/librealsense #8065: L515 not recognized Windows 10, Error 28

chimobb, 2020-12-24:
| | |
|---|---|
|**librealsense**|2.40.0 RELEASE|
|**OS**|Windows|

Please provide a description of the problem

tedtman, 2020-12-30:
Can you provide more details? Not seen in device manager? Windows error message? Is this the same issue as [https://github.com/IntelRealSense/librealsense/issues/7952](https://github.com/IntelRealSense/librealsense/issues/7952)?

chimobb, 2020-12-30:
Ted,
Thank you for contacting, I have solved the problem, changing the USB Port,
it is very sensible at the Port Type, but directly to 3.0 o 3.1 USB Port is
working.

Regards
Joaquin Berenguer

tedtman, 2020-12-30:
@chimobb, great so sounds like we can close this issue, correct?

chimobb, 2020-12-31:
yes, thank you

closed 2021-01-04 by RealSenseSupport