realsenseai/librealsense #8068: arothenberg

arothenberg, 2020-12-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

arothenberg, 2020-12-26:
Also, not the OP subject but what is the ideal way to light a room for depth(only) data. I know to avoid natural light and halogen(?) 
but  should  i get  a studio light? does it make a difference?

MartyG-RealSense, 2020-12-26:
Hi @arothenberg   You can also reset the camera to its factory calibration by using the **Dynamic Calibrator** tool.   you can download it for Windows from the link below.

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

Installing the calibrator tool package also installs a program called **CustomRW**that can be launched from the Windows **Command Prompt** text interface.  The CustomRW program has a factory reset command.

You can find the Command Prompt by typing **command** into the text box at the base of the Windows screen, next to the button in the corner where the PC is shut down.

Once the Command Prompt window is open, you can activate the factory reset function of CustomRW by launching it in the Command Prompt with the command below:

**Intel.RealSense.CustomRW -g**

MartyG-RealSense, 2020-12-26:
In regard to lighting, you do not need to avoid natural light as the **Vision Processor D4** hardware inside the 400 Series cameras is able to auto-adjust to changing lighting conditions very well.  In fact, the 400 Series cameras work even better in strong light.

I would recommand avoiding **fluorescent** lights such as ceiling strip lights that contain heated gas instead of a bulb though, as they can flicker at frequencies that are hard to see with the human eye and introdue noise into the image.   If you cannot avoid such lights in the scene, try setting the camera FPS to a frequency that is close to the operating frequency of the particular fluorescent lights in the scene.  For some lights it might be 60 FOs and for others it may be 30 FPS.

If you do not have strong light present in the scene then you can use the camera's **IR emitter** to cast a semi-random dot pattern over the scene.  The camera can use these dots as a **texture source** to analyze surfaces that the dots are cast upon for depth information.  The laser-based projector in the 400 Series cameras can introduce a noise phenomenon called **laser speckle** though, so if you need to use a dot pattern projection then it may be worth using an external **LED** pattern projector instead.

Alternatively, you could add an external **IR Illuminator** to the scene to provide a light-source to aid the camera's depth analysis.

https://github.com/IntelRealSense/librealsense/issues/2000

arothenberg, 2020-12-26:
Awesome. thanks you for all of that. that did it. 
The manual is a little hinky because it misstates the path to the exe on page 58 jfyi.

I have no natural light in the studio  so I'll try that led. 

Thanks again Marty.

closed 2020-12-26 by arothenberg

arothenberg, 2020-12-30:
One more comment on this. 
That LED helped a lot. I have a dark studio a model and a white back drop. the LED lights got 
rid of a lot of mess from the background  regrading the depth image. 
I think this is the quality of the image I've been looking for. 

Thank you.

MartyG-RealSense, 2020-12-30:
That's great news that LED worked out for you so well.  Thanks very much for the update!   :)