realsenseai/librealsense #8069: compiling "RS Camera" app with L515 support for Android?

PhinPhin, 2020-12-26:
Okay, I don't want to sound like a noob, but in a different post here on Github it was said to use RS Camera on Android with the L515 Lidar-camera you have to compile the app from the source on Github, though I'm kinda stuck there. Which source? I only find examples here, nothing that says obvious "RS Camera app". :)

Who could point me in the right direction?

Oh and yes, I tried the app from the Google PlayStore and it did not work on my Galaxy S9+ with the L515.

RealSenseSupport, 2021-01-28:
@PhinPhin Could you please download the latest apk from the link below and try again? Thanks!
https://dl.bintray.com/intel-realsense/librealsense-dev/com/intel/realsense/librealsense/2.41.0.2657/

RealSenseSupport, 2021-02-03:
@PhinPhin Any update from your side? Looking forward to your reply. Thanks!

PhinPhin, 2021-02-03:
> @PhinPhin Any update from your side? Looking forward to your reply. Thanks!
Oh sorry, i was busy with finding a way to get the cam working on my Surface Pro 3 atleast. I will try the app after work today and will tell you if I have luck!

RealSenseSupport, 2021-02-19:
@PhinPhin Thanks for your reply! Any progress from your side? Thanks!

RealSenseSupport, 2021-03-15:
@PhinPhin Any update from your side? Please note that the issue will be closed in another 7 days if we don't get your reply. Thanks!

RealSenseSupport, 2021-03-31:
@PhinPhin Latest apk is in https://egiintel.jfrog.io/artifactory/librealsense/com/intel/realsense/librealsense/  You can install and run it directly with L515 support. Please try and update. Thanks!

RealSenseSupport, 2021-04-06:
@PhinPhin Sorry that we didn't get response from you for weeks, will close the issue at this point. Please feel free to create another new if you still have questions or issues. Thanks!

closed 2021-04-06 by RealSenseSupport