realsenseai/librealsense #8074: L515 Face Detection can't work when Motion Module opened.

Ralung, 2020-12-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515                    | 
| Firmware Version                   |  01.05.02.00        | 
| Operating System & Version |  Win 10                | 
| Platform                                 | Laptop                 |
| SDK Version                           | v2.40.0                 | 
| Language                              |                              | 
| Segment			        |  others                 | 

### Issue Description

Hello, I have a problem to use "Face Detection : OpenVINO" in RealSense Viewer and other function works fine.
This problem only occur when I open Motion Module and RGB Camera at the same time.
If I only open Depth Sensor and RGB Camera (with or without Face Detection), all of them work fine.
If I open Depth Sensor. RGB Camera without Face Detection and Motion Module. it works fine.
When I first open Motion Module. the Face Detection can't work anymore even if I close Motion Module and only open RGB camera.
The situation will be fixed when I plug L515 in my laptop again.

I am using Razer Blade 15 with i7-1075H, 16GB RAM and RTX 2060 and the RealSense Viewer detected my device as Intel RealSense L515 with USB 3.2 port.
I have tried to downgrade and upgrade the firmware of my L515 but it didn't help.
I found that in Pysical port there is question mark.
[image]

The following pictures are the error code in Viewer
[image]
[image]

If there is any advise, I will be really appreciate! Thanks!

ev-mp, 2020-12-28:
@Ralung hello, this issue was addressed in the latest  v2.41.0, see #7759 for details

RealSenseSupport, 2021-01-03:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

Ralung, 2021-01-09:
Hello,

After flash to the lastest firmware, the problem still exist.
But the error code was changed.
The following is the screenshot of error code.
[image]

RealSenseSupport, 2021-01-10:
Hi @Ralung 

Thank you for the feedback. We have reproduced this (new) issue and are investigating. Will share updates here.

Thanks

RealSenseSupport, 2021-02-28:
Hi @Ralung 

The fix for this has been merged into our dev branch and will also be available as part of the next release.

https://github.com/IntelRealSense/librealsense/pull/8328

Will you be needing any further assistance or can we close this issue?

Thanks

Ralung, 2021-03-01:
Thank you for your help!
I don't need any further assistance in this issue.
Please close this issue

closed 2021-03-01 by RealSenseSupport