realsenseai/librealsense #8078: module 'pyrealsense2' has no attribute 'pipeline'

Lakhsmidhar, 2020-12-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (2.36) | 
| Operating System & Version | Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    | Python 3.7.3  | 
| Platform                                 | Raspberry Pi |
| SDK Version                            |  legacy / 2.36.0 | 
| Language                            |  Python                         | 
| Segment			| Robot               | 

### Issue Description
I`ve installed pyrealsense through this guide https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435.
When i run the example codes in thonny or terminal it's saying "Attribute error module 'pyrealsense2' has no attribute 'pipeline' "

MartyG-RealSense, 2020-12-28:
Hi @Lakhsmidhar   I hope that the link below will be helpful to you.

https://support.intelrealsense.com/hc/en-us/community/posts/360048396854/comments/360012486393

MartyG-RealSense, 2021-01-10:
Case closed due to no further comments received.

closed 2021-01-10 by MartyG-RealSense

sshesh, 2021-03-04:
Late response, posting so it could help others who may be interested to try out. 
I was facing this problem as well (w/ Ubuntu 18.04, Python 3.6.9 with D400 FW version: D400_Series_Minor_FW_5_12_11_0). Below steps worked for me: 
1. Create a virtual env: $python3 -m venv rsvenv
2. Activate: $source rsvenv/bin/activate
3. Upgrade pip:  python3 -m pip install --upgrade pip
4. python3 -m pip install pyrealsense2
5. Install any other packages as required (for me - numpy, scipy, opencv-python, matplotlib, pyglet etc.)

MartyG-RealSense, 2021-03-04:
Thanks so much @sshesh for sharing the method that was successful for you with the RealSense community!   :)