realsenseai/librealsense #8080: Changing resolution in python for L515 camera using SDK2.0

danielroth88, 2020-12-28:
| Required Info                         |              |
|--------------------------------|----------|
| Camera Model                       |L515      | 
| Operating System & Version |  Win10 | 
| Platform                                 | PC        |
| SDK Version                           |  .2.0      | 
| Language                               | Python | 

### Issue Description
When using the python wrapper I want to change to resolution of depth and rgb image data.

I run the following code:
```
import pyrealsense2 as rs 
config = rs.config() 
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
```
but the images doesnt change to the desired size and remain in their default (Depth: 640X480 RGB:1280X720)

what am I doing wrong? are there a limited set of options? if so what are they

Nir-Az, 2020-12-30:
Hi @danielroth88 ,

The code snippet you attached seems fine, but in order to understand the full flow we will need you to attach the rest of the code including how do you use the API to start the pipeline / start sensor stream.

make sure use give the config as a parameter to the pipeline start, example:

import pyrealsense2 as rs 

def frame_arrived(frame):
    pass

config = rs.config() 
pipe = rs.pipeline()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
pipe.start(config, frame_arrived)
....
pipe.stop()

danielroth88, 2020-12-30:
I didn't connect it to the pipleline (through pipeline.start() ) - thanks!

Nir-Az, 2020-12-30:
Great! , good luck 👍

closed 2021-01-05 by ev-mp