realsenseai/librealsense #8083: Cannot install pyrealsense

yeric1789, 2020-12-30:
Cannot Install pyrealsense
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435i  | 
| Firmware Version                   | 5.12.09.00 | 
| Operating System & Version |  Raspbian GNU/Linux 10 (buster), Release version 10                                    | 
| Platform                                 | Raspberry Pi 4 4GB|
| SDK Version                            | 2.40 |
| Language                            |  python                      | 

Hi, 
I am attempting to set up a raspberry pi system similar to the one outlined in this file.  [https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras](url). I have already solved multiple problems regarding streaming video however in order to set up a client-server system between my raspberry pi 4's, I need to install pyrealsense on each of the raspberry pi's. However if it is already installed my python version 3.7 doesn't detect it. 
I have downloaded the pre-built image found on [https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras](url)
Please see the errors I get from the command line below
Error when trying to build. For installing pyrealsense:

```
realsense@realsense:~/librealsense/build $ cmake .. -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=$(which python3)
-- Internet connection identified
CMake Deprecation Warning at CMakeLists.txt:10 (MESSAGE):
  FORCE_LIBUVC, FORCE_WINUSB_UVC and ANDROID_USB_HOST_UVC are deprecated, use
  FORCE_RSUSB_BACKEND instead

-- Info: REALSENSE_VERSION_STRING=2.34.1
-- Setting Unix configurations
-- using RS2_USE_LIBUVC_BACKEND
-- LIVE555: File exists, delete it to download again (/home/realsense/librealsense/build/third-party/live555/live555-latest.tar.gz)
-- LIVE555: Unpacking source
-- LIVE555: Source unpacked
-- pybind11 v2.2.1
-- GLFW 3.3 not found; using internal version
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR) 
-- Using X11 for window creation
-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
-- Unable to find apriltag library, skipping pose-apriltag example
-- Building with TM2
-- Fetching recommended firmwares:
-- D4XX_RC_VERSION: 5.12.3.0
-- D4XX_FW_VERSION: 5.12.3.0
-- SR3XX_FW_VERSION: 3.26.1.0
-- T26X_FW_VERSION: 0.2.0.951
-- http://realsense-hw-public.s3.amazonaws.com/Releases/RS4xx/FW/D4XX_FW_Image-5.12.3.0.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for D4XX_FW_Image-5.12.3.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/RS4xx/FW/D4XX_RC_Image-5.12.3.0.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for D4XX_RC_Image-5.12.3.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for SR3XX_FW_Image-3.26.1.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.2.0.951/target-0.2.0.951.mvcmd
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for target-0.2.0.951.mvcmd
-- Configuring done
-- Generating done
-- Build files have been written to: /home/realsense/librealsense/build
realsense@realsense:~/librealsense/build $ make -j1 && sudo make install
[  0%] Performing update step for 'libjpeg-turbo'
Current branch master is up to date.
[  0%] Performing configure step for 'libjpeg-turbo'
-- CMAKE_BUILD_TYPE = Release
-- VERSION = 2.0.91, BUILD = 20200412
-- 32-bit build (arm)
-- CMAKE_INSTALL_PREFIX = /home/realsense/librealsense/build/libjpeg-turbo
-- CMAKE_INSTALL_BINDIR = bin (/home/realsense/librealsense/build/libjpeg-turbo/bin)
-- CMAKE_INSTALL_DATAROOTDIR = share (/home/realsense/librealsense/build/libjpeg-turbo/share)
-- CMAKE_INSTALL_DOCDIR = share/doc/libjpeg-turbo (/home/realsense/librealsense/build/libjpeg-turbo/share/doc/libjpeg-turbo)
-- CMAKE_INSTALL_INCLUDEDIR = include (/home/realsense/librealsense/build/libjpeg-turbo/include)
-- CMAKE_INSTALL_LIBDIR = lib (/home/realsense/librealsense/build/libjpeg-turbo/lib)
-- CMAKE_INSTALL_MANDIR = share/man (/home/realsense/librealsense/build/libjpeg-turbo/share/man)
-- Shared libraries enabled (ENABLE_SHARED = 1)
-- Static libraries enabled (ENABLE_STATIC = 1)
[... 97 lines cut ...]
    copying pyrealsense/extlib.py -> build/lib.linux-armv7l-3.7/pyrealsense
    running build_ext
    building 'pyrealsense.rsutilwrapper' extension
    creating build/temp.linux-armv7l-3.7
    creating build/temp.linux-armv7l-3.7/pyrealsense
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Ipyrealsense -I/usr/lib/python3/dist-packages/numpy/core/include -I/usr/local/include/librealsense -I/usr/include/python3.7m -c pyrealsense/rsutilwrapper.cpp -o build/temp.linux-armv7l-3.7/pyrealsense/rsutilwrapper.o
    In file included from /usr/lib/python3/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1822,
                     from /usr/lib/python3/dist-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                     from /usr/lib/python3/dist-packages/numpy/core/include/numpy/arrayobject.h:4,
                     from pyrealsense/rsutilwrapper.cpp:654:
    /usr/lib/python3/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
     #warning "Using deprecated NumPy API, disable it with " \
      ^~~~~~~
    pyrealsense/rsutilwrapper.cpp:657:10: fatal error: rs.h: No such file or directory
     #include "rs.h"
              ^~~~~~
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-z0yp_4dy/pyrealsense/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-x0okz7g3/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-install-z0yp_4dy/pyrealsense/
```
Is there some way to solve this problem. Any help you could provide would be much appreciated. I am quite new to this forum so any pointers on making a clean build would also be appreciated (ie uninstalling an old build and reinstalling librealsense).
Please let me know if you need any more details.
Thanks.

MartyG-RealSense, 2020-12-30:
Hi @yeric1789  The **pip install** method will not work on a Raspberry Pi, as it has an **Arm** processor and the pyrealsense2 pip packages are designed for x86 / x64 processors like those in desktop / laptop computers.  The Python wrapper should be built from **source code** on Arm devices.

Given that you are using Raspbian, I would recommend building librealsense and Python together on your Pi 4 using the popular **Acrobotic** guide.  The structure of this guide is similar to  Intel's own Raspbian guide for installing librealsense.

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

Once you have librealsense and Python installed on the Pi then this can act as a stable foundation to start constructing your own networking system around it.

yeric1789, 2021-01-04:
I was able to get pyrealsense2 on the RPI 4 however I believe I ran into a similar issue as stated in issue #6820 , where I can import pyrealsense however when I attempt to run code it states
`module pyrealsense2 has no attribute config` 
The same goes for the attribute context. I am quite confused reading through these issues as the solution to this problem. I already installed pyrealsense is there a way to modify it and or add the attributes. Any guidance would be much appreciated thanks.

MartyG-RealSense, 2021-01-04:
I think that **ModuleNotFoundError: No module named 'pyrealsense'** is more significant as a possible cause of your problem on Pi 4.

I notice that it states that 'pyrealsense' is not found.  It is normally 'pyrealsense2' in the error message.  Could you tell me what method you used to install the Python wrapper please?  I would like to confirm that you did not install pyrealsense 2.2 (an older pyrealsense that is incompatible with RealSense SDK 2.0).

https://pypi.org/project/pyrealsense/

yeric1789, 2021-01-04:
The method I used to install pyrealsense was outlined in the article [https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435](url)

I was able to eventually get pyrealsense working however it only works when I insert the specified python file into directory path `/home/librealsense/build/wrappers/python/`

Is there any way to apply this utility across the entire home directory?

MartyG-RealSense, 2021-01-05:
I would expect that the defining of the **PYTHONPATH** variable would avoid the need to place files beside your script.  Can you confirm please that you inputted the command below after setting the PYTHONPATH in order to confirm the change made to the path:

**source ~/.bashrc**

[image]

yeric1789, 2021-01-05:
I do have the text
`export PYTHONPATH=$PYTHONPATH:/usr/local/lib`

However after doing `source ~/.bashrc` it seems to now to work.
Thanks for your help.

yeric1789, 2021-01-05:
Wait scratch that I am still getting the same error outside

yeric1789, 2021-01-05:
I have applied the change you specified above and still no change. It doesn't seem to be recognizing pyrealsense unless I am in the python wrapper folder. Could it be that PYTHONPATH :/usr/local/lib automatically references python 2 instead of python 3? So when I am modifying the path it could be looking for pyrealsense in the wrong python distribution?

MartyG-RealSense, 2021-01-05:
Researching this problem reminded me that installing pyrealsense2 on Raspberry Pi can be more complicated than other devices.  Of the past cases of this kind of problem that I have handled, I believe that the link below provides the best advice.

https://support.intelrealsense.com/hc/en-us/community/posts/360048396854/comments/360012486393

closed 2021-01-13 by yeric1789