realsenseai/librealsense #8086: How to get the center distance with depth camera.

junhui519, 2020-12-30:
| Required Info                         |                   distance data                            |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.03.00 | 
| Operating System & Version | Ubuntu 18.04| 
| Kernel Version (Linux Only)    |  5.4.0-58-generic | 
| Platform                                 | Ros |
| SDK Version                            |  { legacy / 2.41.0 }                          | 
| Language                            |  C#                        | 
| Segment			|  Robot                        | 

### Issue Description
Hi, I'm trying to get center distance data using L515. So I run rs_camera.launch which I downloaded from github to find and tried to find a topic which has distance data. But realized that I can't get the distance right from it. So I decided to use librealsense examples, hello-realsense.
But I have no idea about how to run the rs-hello-realsense.cpp. I followed instruction and did Cmake. But what should I do next? like rosrun? I'm so beginner about this. Please answer this and let me know where can I learn something about this.

RealSenseSupport, 2021-01-06:
Hi @junhui519 

For any of our RealSense devices including L515, I would start by using our RealSense Viewer (https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer) tool in order to get familiar with the device and the capabilities that it has to offer.  We provide an executable such that you wouldn't have to build it directly but you can run the executable and get started from there.

Here is a link to the executable:  https://github.com/IntelRealSense/librealsense/releases/download/v2.41.0/Intel.RealSense.Viewer.exe

Here are some other links as well regarding L515 that can help get started:
L515 User Guide -  https://www.intelrealsense.com/download/13061/
Optimizing for distance - https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/
List of other documents - https://dev.intelrealsense.com/docs/lidar-l500

Hope this helps

junhui519, 2021-01-07:
Thank you for your information! But I was already using realsense-viewr and it works well. But what I want to do is to get some data from some topics, so that I want run librealsense's examples like rs-measure.cpp. I usually run the cpp files, locating it in my workspace, doing catking_make and rosrun. But this time, I don't know how to run librealsense's examples. Please let me know how to use these.

closed 2021-01-07 by junhui519