realsenseai/librealsense #8088: Power Cycling the Realsense.

GrubbyHalo, 2020-12-30:
Has anyone had to resort to power cycling the realsense D435 cameras ? (i.e. due to it locking up, freezing etc) 
I have a realsense running on a raspberry pi 3/4 with a custom HAT on top of the pi that provides power to the raspberry pi. The hat is only capable of hard resetting the pi (via the RUN pin). this only resets the SoC. I am wondering if this is sufficient in case the camera locks up.

MartyG-RealSense, 2020-12-30:
Hi @GrubbyHalo   Whilst the camera can be reset with a hard or soft reboot, it is also possible to reset only the camera hardware without resetting the entire computer if a freeze in the camera streaming occurs.  

This can be done with the **Hardware Reset** function in the RealSense Viewer program (under the **More** option of the Viewer's options side-panel) or from within program scripts by invoking the RealSense SDK's **hardware_reset()** instruction.   This instruction can be set up to auto-trigger if certain conditions that suggest that a problem has occurred are met (for example, if more than 50% of the pixels on an image have zero depth).

Another means of resetting the camera without restarting the computer is to run a USB port reset script, which resets the entire port instead of just the camera specifically.

There are some error situations that can be corrected with a camera or USB port reset, and others that require a full shutdown and reboot to correct the problem.  The need to perform a full shutdown is rare though in my experience, and a camera-only reset (whether through hardware reset or unplugging the camera and re-inserting it into the USB port) often corrects a freeze.

GrubbyHalo, 2020-12-30:
Thanks for the speedy response. From your answer i am reading that it's unlikely for the camera to require a hardware reset via unplugging the and re-inserting the camera into the usb port but its not impossible for it to need this. 

I have hundreds of D435's installed at remote locations that currently are able to completely power cycle should an issue occur. However due to certain constraints, its difficult for our next version of our HAT to cut power to the PI. That's why i was hoping that there would be some kind of guarantee that the D435 would never require power cycle. But based on your response i think i might need to revise our next version of the HAT.

MartyG-RealSense, 2020-12-30:
Are you using a Poe HAT like the one in the Pi store?

https://www.raspberrypi.org/products/poe-hat/

Intel's open-source ethernet networking project recommends one of these but states that an external power source via Raspberry Pi USB-C connector can be used instead.  Would an external USB-C power source be practical for your project instead of a custom-made HAT?

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-1-list-of-components

GrubbyHalo, 2020-12-30:
@MartyG-RealSense our current HAT  supplies power to the pi. The power to the pi can be disconnected while still having the rest of the HAT powered. (we switch the power via a mosfet which is in turn controlled via a small PIC micro on the hat. This Pic switches the power if it doesn't get a response from the pi within a certain time. I. e almost like a watchdog).

The poe HAT which is a requirement for us in some cases does not have the facility for us to independently control power to the pi.

MartyG-RealSense, 2020-12-30:
Thanks very much @GrubbyHalo for the information about your HAT.  

The 400 Series cameras are able to run indefinitely as long as they remain within recommended temperature tolerances.  If a problem does occur during a long-run session then if the temperature is within tolerances, the problem is likely related to a non-camera element of the setup such as a glitch in the USB cable / port, the computer hardware or the operating system.

If those components are stable then you can also reduce the likelihood of the camera stream freezing by trying to ensure that the camera is not pointed directly at the sun.  This can cause the IR sensor to become saturated - The fix for this is to remove the camera from direct view of the light (by moving the camera or covering the lenses) to give the camera's auto-exposure a chance to adjust and recover, or to reset the camera if auto-adjustment is not successful.

If there are times of day when direct exposure to the sun may be unavoidable, you could consider overlaying a physical **optical filter** over the top of the camera's lenses.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400

GrubbyHalo, 2020-12-30:
@MartyG-RealSense thank you for input. It has provided me with valuable information that will help us determine how to implement our HAT solution.

closed 2020-12-30 by GrubbyHalo