realsenseai/librealsense #8089: Zoom

Imyyy, 2020-12-30:
Hi, 
I am using the RealSense SDK viewer in windows 10. The sotware zooms in to a very large resolution on the screen, so I am unablet o see most of the information. I have not found a way to adjust it in windows zoom / display / magnification settings. 
Is there somethingI can do to fix this? 

Thanks in advance, 
Imy

| | |
|---|---|
|**librealsense**|2.40.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|745412070502|
|**Firmware Version**|05.12.09.00|
|**Advanced Mode**|NO|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|804113020928|
|**Firmware Update Id**|804113020928|

Please provide a description of the problem

MartyG-RealSense, 2020-12-30:
Hi @Imyyy   I would first recommend toggling fullscreen mode.  This can be done in the Viewer's **Settings** options, but if the interface is hard to read / use because of the zoom then you can also toggle it by pressing the **F8** keyboard key.

Imyyy, 2020-12-30:
@MartyG-RealSense that worked!Thanks so much for your time

MartyG-RealSense, 2020-12-30:
Great to hear - thanks for the update  :)

closed 2021-01-06 by MartyG-RealSense