realsenseai/librealsense #8090: Error compiling SDK in Apple Silicon

hehengxiang0, 2020-12-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {MacOS  | 
| Platform                                 | Apple Silicon  |
| SDK Version                            |  { legacy / 2.<41>.<0> }                          | 
| Language                            |  {C }                          | 
| Segment			|  {others }                          | 

### Issue Description
Error compiling SDK in Apple Silicon

make -j2

In file included from /Users/hehengxiang/Movies/software/librealsense/src/proc/color-formats-converter.cpp:31:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.0/include/cpuid.h:11:2: error: 
      this header is for x86 only
#error this header is for x86 only
 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.0/include/cpuid.h:271:5: error: 
      invalid output constraint '=a' in asm
    __cpuid(__leaf, __eax, __ebx, __ecx, __edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.0/include/cpuid.h:236:11: note: 
      expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.0/include/cpuid.h:286:5: error: 
      invalid output constraint '=a' in asm
    __cpuid(__leaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.0/include/cpuid.h:236:11: note: 
      expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.0/include/cpuid.h:300:5: error: 
      invalid output constraint '=a' in asm
    __cpuid_count(__leaf, __subleaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.0/include/cpuid.h:243:11: note: 
      expanded from macro '__cpuid_count'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Users/hehengxiang/Movies/software/librealsense/src/proc/color-formats-converter.cpp:33:5: error: invalid output constraint
      '=a' in asm
    __cpuid_count(info_type, 0, info[0], info[1], info[2], info[3]);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.0/include/cpuid.h:243:11: note: 
      expanded from macro '__cpuid_count'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
[ 63%] Building CXX object CMakeFiles/realsense2.dir/src/proc/depth-formats-converter.cpp.o
5 errors generated.
make[2]: *** [CMakeFiles/realsense2.dir/src/proc/color-formats-converter.cpp.o] Error 1

MartyG-RealSense, 2020-12-30:
Hi @hehengxiang0   The link below leads to a discussion where somebody else had a similar problem in December 2020 with installing librealsense on a new M1 processor based Mac. 

http://5.9.10.113/65154758/intel-real-sense-d435-use-on-m1-mac

It is likely that librealsense has not been tested on an M1 Mac at the time of writing this due to the newness of the technology.  I will therefore seek advice on the subject from Intel.   Thanks for your patience!

MartyG-RealSense, 2020-12-31:
Hi again @hehengxiang0   Intel responded that they will look into it.

hehengxiang0, 2021-01-01:
> Hi again @hehengxiang0 Intel responded that they will look into it.

Thank you！
Waiting for your good news～

dyamized, 2021-04-24:
I have compiled the library for Apple Silicon by adding "|| defined(__aarch64__)" to the line before "bool has_avx() { return false; }":

```
#if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__)) || defined(__aarch64__)

bool has_avx() { return false; }
```

And the library appears to be working properly in my code.

MartyG-RealSense, 2021-04-25:
Thanks very much @dyamized for sharing the method that worked for you!   I will pass your information to Intel.

RezoGreywords, 2021-04-25:
> I have compiled the library for Apple Silicon by adding `|| defined(__aarch64__)` to the line before "bool has_avx() { return false; }":
>
> ```
> #if defined (ANDROID) || (defined (__linux__) && [...]

hi, may I ask where (which file?) can I find and modify this?

I am using macOS BigSure 11.1 with Xcode 12.4(installed via appstore) and Apple clang version 12.0.0.
I have followed instruction on https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md
with extra steps (sudo chown -R $(whoami) /usr/local/  to get vulkan-sdk properly installed, 
git clone https://github.com/IntelRealSense/librealsense.git and cd librealsense...etc

now I am stocking on the last procedure of step 4:
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false

at beginning cmake returned pthread.h  not found error, I tried adding extra sets in CMakeLists.txt as instructed from #8250

> set(CMAKE_THREAD_LIBS_INIT "-lpthread")
> set(CMAKE_HAVE_THREADS_LIBRARY 1)
> set(CMAKE_USE_WIN32_THREADS_INIT 0)
> set(CMAKE_USE_PTHREADS_INIT 1)
> set(THREADS_PREFER_PTHREAD_FLAG ON)

and now I am getting these errors from cmake :

CMake Error at /opt/homebrew/Cellar/cmake/3.20.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.20.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.20.1/share/cmake/Modules/FindThreads.cmake:238 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  examples/measure/CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
See also "/users/Rezo/librealsense/build/CMakeFiles/CMakeOutput.log".
See also "/users/Rezo/librealsense/build/CMakeFiles/CMakeError.log".

  CMakeError.log reads 

> Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
> Compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ 
> Build flags: 
> Id flags:  
> 
> The output was:
> 1
> ld: library not found for -lc++
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> 
> 
> Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
> Compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc 
> Build flags: 
> Id flags:  
> 
> The output was:
> 1
> ld: library not found for -lSystem
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> 
> 
> Determining if the include file pthread.h exists failed with the following output:
> Change Dir: /users/Rezo/librealsense/build/CMakeFiles/CMakeTmp
> 
> Run Build Command(s):/usr/bin/make -f Makefile cmTC_3597a/fast && /Applications/Xcode.app/Contents/Developer/usr/bin/make  -f CMakeFiles/cmTC_3597a.dir/build.make CMakeFiles/cmTC_3597a.dir/build
> Building C object CMakeFiles/cmTC_3597a.dir/CheckIncludeFile.c.o
> /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc   -pedantic -g -D_DEFAULT_SOURCE   -mfpu=neon -mfloat-abi=hard -ftree-vectorize -latomic -pthread  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -fPIE -MD -MT CMakeFiles/cmTC_3597a.dir/CheckIncludeFile.c.o -MF CMakeFiles/cmTC_3597a.dir/CheckIncludeFile.c.o.d -o CMakeFiles/cmTC_3597a.dir/CheckIncludeFile.c.o -c /users/Rezo/librealsense/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
> clang: warning: -latomic: 'linker' input unused [-Wunused-command-line-argument]
> clang: warning: argument unused during compilation: '-mfpu=neon' [-Wunused-command-line-argument]
> clang: warning: argument unused during compilation: '-mfloat-abi=hard' [-Wunused-command-line-argument]
> Linking C executable cmTC_3597a
> /opt/homebrew/Cellar/cmake/3.20.1/bin/cmake -E cmake_link_script CMakeFiles/cmTC_3597a.dir/link.txt --verbose=1
> /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc    -pedantic -g -D_DEFAULT_SOURCE   -mfpu=neon -mfloat-abi=hard -ftree-vectorize -latomic -pthread  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/cmTC_3597a.dir/CheckIncludeFile.c.o -o cmTC_3597a 
> ld: library not found for -latomic
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> make[1]: *** [cmTC_3597a] Error 1
> make: *** [cmTC_3597a/fast] Error 2

I am wondering if it is missing the  -lc++ and -lSystem
I have tried to port install ld64-xcode and the problem still persists.

Thanks in advance for anyone helping.

dyamized, 2021-04-25:
I didn't try to install Vulkan.

I was following the simpler instructions here:
https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk

The file I changed is the original one listed in the bug report with the
error: librealsense/src/proc/color-formats-converter.cpp

wangyuan02605, 2021-04-29:
i update the code in  color-formats-converter.cpp file ,but i find a new error

ld: library not found for -latomic
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [CMakeFiles/realsense2.dir/build.make:2277: librealsense2.2.44.0.dylib] Error 1

dyamized, 2021-04-30:
I don't think you followed the build instructions on this page:
https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk

Since these instructions create an Xcode project.

wangyuan02605, 2021-04-30:
thanks i can build it in xcode ,but can you tell me the next (such as  run a demo),i want to use it in the python3

wangyuan02605, 2021-04-30:
ohh i can run the demo!!! ,just click the file in debug folder

cpkcpkcpk, 2021-05-09:
Hi, I'm still getting errors building in Xcode on Apple Silicon.

After entering this line `cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -G Xcode`, I get the following error:

`CMake Error at /opt/homebrew/Cellar/cmake/3.20.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY
  OPENSSL_INCLUDE_DIR)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.20.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.20.2/share/cmake/Modules/FindOpenSSL.cmake:570 (find_package_handle_standard_args)
  CMake/external_libcurl.cmake:56 (find_package)
  CMake/global_config.cmake:93 (include)
  CMakeLists.txt:44 (global_set_flags)`

I can create the Xcode project by adding the line `export OPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl` and editing the file that @dyamized posted above, but building fails with these openssl errors relating to libcurl:

`PhaseScriptExecution Generate\ libcurl/src/libcurl-stamp/$(CONFIGURATION)$(EFFECTIVE_PLATFORM_NAME)/libcurl-configure /Users/charliekemp/Documents/librealsense/build/librealsense2.build/Debug/libcurl.build/Script-EEA27806F0687504DA280E8B.sh (in target 'libcurl' from project 'librealsense2')
    cd /Users/charliekemp/Documents/librealsense
    /bin/sh -c /Users/charliekemp/Documents/librealsense/build/librealsense2.build/Debug/libcurl.build/Script-EEA27806F0687504DA280E8B.sh

-- The C compiler identification is AppleClang 12.0.5.12050022
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- curl version=[7.75.0-DEV]
-- Performing Test HAVE_SOCKADDR_IN6_SIN6_ADDR
-- Performing Test HAVE_SOCKADDR_IN6_SIN6_ADDR - Success
-- Performing Test HAVE_SOCKADDR_IN6_SIN6_SCOPE_ID
-- Performing Test HAVE_SOCKADDR_IN6_SIN6_SCOPE_ID - Success
-- Found Perl: /usr/bin/perl (found version "5.30.2") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Looking for dlopen in ;
-- Looking for dlopen in ; - found
-- Looking for connect in socket;
-- Looking for connect in socket; - not found
-- Looking for gethostbyname in c
-- Looking for gethostbyname in c - found
-- Looking for gethostname
-- Looking for gethostname - found
CMake Error at /opt/homebrew/Cellar/cmake/3.20.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY
  OPENSSL_INCLUDE_DIR)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.20.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.20.2/share/cmake/Modules/FindOpenSSL.cmake:570 (find_package_handle_standard_args)
  CMakeLists.txt:387 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/charliekemp/Documents/librealsense/build/libcurl/src/libcurl-build/CMakeFiles/CMakeOutput.log".
See also "/Users/charliekemp/Documents/librealsense/build/libcurl/src/libcurl-build/CMakeFiles/CMakeError.log".
Command PhaseScriptExecution failed with a nonzero exit code`

Does anyone have any ideas how to fix these errors? I'm guessing I need to add some lines to CMakeLists.txt, but I'm not sure what exactly.

Thanks!

c-mendoza, 2021-05-10:
Hey @cpkcpkcpk I was getting the same error in building libcurl, fixed it by modifying `CMake/external_libcurl.cmake`. 

After line 35, I added:

`-DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl`

Basically it seems that the internal cmake call doesn't get the flags passed along in the main cmake call, so it doesn't know where OpenSSL is to be found.

Got it to compile and run realsense_viewer, connected a camera and runs as it's supposed to, so yay for that.

MartyG-RealSense, 2021-05-10:
Thanks so much @c-mendoza for sharing with @cpkcpkcpk and the RealSense community the solution that worked for you!

c-mendoza, 2021-05-10:
Looks like something got built for x86... try deleting the CMake cache folders, generating the project again and then building again in Xcode.

alifeinbinary, 2021-05-20:
Thanks @MartyG-RealSense. This thread got me to a completed build. 

Investigating a little further...
`which openssl` on my system revealed that MacOS defaults to using openssl at `/usr/bin/openssl`, which is actually LibreSSL (version 2.8.3 in my case). 

Using `brew unlink openssl && brew link openssl --force` resulted in a message instructing my to export the following variables in order for openssl, not LibreSSL, to be used during builds. It was this particular detail that got me to ta completed build using `cmake .. -DBUILD_PYTHON_BINDINGS:bool=true -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false`  
```
export PKG_CONFIG_PATH="/usr/local/opt/openssl@1.1/lib/pkgconfig"
export CPPFLAGS="-I/usr/local/opt/openssl@1.1/include"
export LDFLAGS="-L/usr/local/opt/openssl@1.1/lib"
```

RealSenseSupport, 2021-06-11:
I think this thread can be closed for now since a solution has been provided. A ticket had been filed internally for expanding this support in future SDK.
Thank you all for the contribution and efforts to enable RealSense on Apple M1.

closed 2021-06-11 by RealSenseSupport

jalapatip, 2021-06-30:
@MartyG-RealSense, @c-mendoza and @dyamized

I could successfully build the project but am not able to run realsense-viewer. Initially I got the below error and I fixed it by adding 'set(CMAKE_MACOSX_RPATH 1)' to the CMakeLists.txt file.

Initial error:
MACOSX_RPATH is not specified for the following targets: realsense2-gl

Current error:
'zsh: command not found: realsense_viewer'

Could you please help me fix this.

gschian0, 2021-09-02:
HI, I"m having an issue install sdk on apple silicon ... my issue is detailed on this issue any help would be appreciated! [https://github.com/IntelRealSense/librealsense/issues/9709](url)

rlabs-oss, 2021-09-06:
-- Info: REALSENSE_VERSION_STRING=2.49.0
-- Setting Unix configurations
-- Building libcurl enabled
CMake Error at /opt/homebrew/Cellar/cmake/3.21.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY
  OPENSSL_INCLUDE_DIR)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.21.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.21.2/share/cmake/Modules/FindOpenSSL.cmake:574 (find_package_handle_standard_args)
  CMake/external_libcurl.cmake:56 (find_package)
  CMake/global_config.cmake:93 (include)
  CMakeLists.txt:44 (global_set_flags)

-- Configuring incomplete, errors occurred!

openssl@1.1 1.1.1l is already installed and up-to-date.
To reinstall 1.1.1l, run:
  brew reinstall openssl@1.1

brew is a total mess.

gschian0, 2021-09-06:
hey! thanks @rlabs-oss !! I'm further than I was... but now I'm getting this error after running `make -j2`
Is there a way to build this for Xcode and run it like that?  Thanks for any help in advance!

`n file included from /Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/motion-transform.cpp:4:
/Users/gennaroschiano/Documents/GitHub/librealsense/src/ds5/ds5-options.h:344:22: warning: 'is_read_only' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
        virtual bool is_read_only() const { return _sensor && _sensor->is_opened(); }
                     ^
/Users/gennaroschiano/Documents/GitHub/librealsense/src/core/options.h:27:22: note: overridden virtual function is here
        virtual bool is_read_only() const { return false; }
                     ^
In file included from /Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/motion-transform.cpp:4:
/Users/gennaroschiano/Documents/GitHub/librealsense/src/ds5/ds5-options.h:367:22: warning: 'is_read_only' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
        virtual bool is_read_only() const { return _sensor && _sensor->is_opened(); }
                     ^
/Users/gennaroschiano/Documents/GitHub/librealsense/src/core/options.h:27:22: note: overridden virtual function is here
        virtual bool is_read_only() const { return false; }
                     ^
In file included from /Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/motion-transform.cpp:5:
In file included from /Users/gennaroschiano/Documents/GitHub/librealsense/src/ds5/ds5-motion.h:6:
In file included from /Users/gennaroschiano/Documents/GitHub/librealsense/src/ds5/ds5-device.h:14:
/Users/gennaroschiano/Documents/GitHub/librealsense/src/fw-update/fw-update-device-interface.h:27:102: warning: adding 'size_t' (aka 'unsigned long') to a string does not append to the string [-Wstring-plus-int]
                throw std::runtime_error("Firmware binary image might be corrupted - size is only: " + fw_image_size);
                                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/Users/gennaroschiano/Documents/GitHub/librealsense/src/fw-update/fw-update-device-interface.h:27:102: note: use array indexing to silence this warning
                throw std::runtime_error("Firmware binary image might be corrupted - size is only: " + fw_image_size);
                                                                                                     ^
                                         &                                                           [              ]
4 warnings generated.
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
make: *** [all] Error 2`

rlabs-oss, 2021-09-06:
xcode - not that Im aware of. - brew with perquisite libraries ideally packaged to brew install (apple silicon pending)  ... otherwise packages installed via git clone will spew files and cause inconsistencies. e.g installing to /use/local and not /opt/homebrew most dont even honor the cmake prefix. I'd delete everything in librealsense, git clone it again and start again. At one point I did setup a terminal to start under Rosetta as x86 and brew install that way - but thats just asking for more trouble once you want to use brew to build native apps. They dont play together.

gschian0, 2021-09-06:
well.. I tried the Xcode thing.. and it looks like the issue is the **cpuid.h** file.
It hangs up on this part of the code ... `#if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__))

bool has_avx() { return false; }

#else

#ifdef _WIN32
#include <intrin.h>
#define cpuid(info, x)    __cpuidex(info, x, 0)
#else
#include <cpuid.h>
void cpuid(int info[4], int info_type) {
    __cpuid_count(info_type, 0, info[0], info[1], info[2], info[3]);
}
//#endif

bool has_avx()
{
    int info[4];
    cpuid(info, 0);
    cpuid(info, 0x80000000);
    return (info[2] & ((int)1 << 28)) != 0;
}

#endif`

gschian0, 2021-09-06:
just to be clear I also added these lines to my **global_config.cmake** file 

`set(CMAKE_THREAD_LIBS_INIT "-lpthread")
set(CMAKE_HAVE_THREADS_LIBRARY 1)
set(CMAKE_USE_WIN32_THREADS_INIT 0)
set(CMAKE_USE_PTHREADS_INIT 1)
set(THREADS_PREFER_PTHREAD_FLAG ON)`

shadowgroundz, 2021-09-10:
> Hey @cpkcpkcpk I was getting the same error in building libcurl, fixed it by modifying `CMake/external_libcurl.cmake`.
>
> After line 35, I added:
>
> `-DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl` [...]

Hi buddy. I succeed build with your help. But I am not familiar with xcode.  After succeed how you run `realsense_viewer`?

MartyG-RealSense, 2021-09-10:
Hi @shadowgroundz   Could you check whether the Unix files to run realsense-viewer are located in **/usr/local/Cellar/librealsense/**

JoeFirmament, 2021-09-22:
> Hi again @hehengxiang0 Intel responded that they will look into it.

Any info update?

MartyG-RealSense, 2021-09-22:
Hi @JoeFirmament  There is not currently an official guide to installing on M1 Mac, though solutions for most of the problems that may be encountered during an M1 installation have been shared by RealSense users on this discussion and other discussions linked to from this page.

JoeFirmament, 2021-09-22:
> Hi @JoeFirmament There is not currently an official guide to installing on M1 Mac, though solutions for most of the problems that may be encountered during an M1 installation have been shared by RealSense users on this discussion and other discussions linked to from this page.

Thx a lot for ur feedback @MartyG-RealSense . I've already solve it by reading this discussion again and again.

Obrepal, 2022-02-22:
> latomi

How did solve latomic issue?

MartyG-RealSense, 2022-04-24:
Hi everyone, I just wanted to highlight that the **brew install librealsense** method of installing librealsense on Mac now lists support for MacOS Monterey and for Apple Silicon M1.

https://formulae.brew.sh/formula/librealsense

Obrepal, 2022-04-25:
Great news!
What step was skipped if there is no module named 'pyrealsense2'?
installed via brew

MartyG-RealSense, 2022-04-25:
Hi @Obrepal  If the brew method was used then the pyrealsense2 wrapper would likely not be included in the build of librealsense, as it is usually not included in librealsense packages or only built from source code at the same time as building librealsense if including the build flag **-DBUILD_PYTHON_BINDINGS:bool=true** in the CMake build instruction.

In that case, you may be able to use the pre-built pyrealsense2 package for MacOS by **cansik** that is referenced in the lightbuzz guide to install pyrealsense2 on your Mac once librealsense has been installed by brew.

https://github.com/cansik/pyrealsense2-macosx/tree/apple-silicon

Obrepal, 2022-04-30:
Ok, did somebody handled this issue?

```
pipeline_profile = config.resolve(pipeline_wrapper)
RuntimeError: No device connected
```

lsusb:
```
Bus 000 Device 001: ID 8086:0b07 Intel Corporation Intel(R) RealSense(TM) Depth Camera 435 
Bus 000 Device 000: ID 8086:0b07 Intel Corporation USB 3.1 Bus 
```
I looked at other posts, but due to different origin of problem they were not much helpful