realsenseai/librealsense #8091: Memory leak on restarting streams

Evlasey, 2020-12-30:
**SDK Version:** v2.38.1.2223
**OS:** Windows 10 Home 64 bit
**Platform:** Laptop (_Intel Core i7-10510U, 16G RAM, NVIDIA GeForce MX250 2G_)
**Language:** C++ (_Visual Studio 2019_)
**Segment:** Desktop 
**Camera Model:** D435i
**Firmware Version:** 05.12.07.100

### Issue Description

Recently we noticed some memory issues in our project. Each time we make restarts in our camera controller to change configuration, we face memory leak. We also tested our use case on different versions of RealSense SDK, but the problem is still there.

### Sample to reproduce the Issue
_main.cpp_:
<details> 

```cpp
#include <librealsense2/rs.hpp>
#include <iostream>

static const int restarts = 200;
static int restarts_count = 0;
static const int frames_to_restart = 5;

// Global RS variables
rs2::context g_context;
std::shared_ptr<rs2::pipeline> g_pipeline = std::make_shared<rs2::pipeline>();
std::shared_ptr<rs2::config> g_config = std::make_shared<rs2::config>();

class StreamDataController {

public:

    void init() {

        auto devices = g_context.query_devices();
        
        if (devices.size() == 0)
            throw "Realsense2DepthProvider: Can't enumerate Realsense2 camera";

        g_config->enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
    }

    void release() {
        g_config->disable_all_streams();
    }

    void start() {
        g_pipeline->start(*g_config);
    }

    void stop() {
        g_pipeline->stop();
    }

    void handle() {
        int frame_count = 0;
        while (frame_count < frames_to_restart) {
            auto frames = g_pipeline->wait_for_frames();

            auto depth_frame = frames.get_depth_frame();

            // Get the depth frame's dimensions
            float width = depth_frame.get_width();
            float height = depth_frame.get_height();

            // Query the distance from the camera to the object in the center of the image
            float dist_to_center = depth_frame.get_distance(width / 2, height / 2);

            std::cout << "<Depth sensor> frame=" << frame_count << "\tThe camera is facing an object " << dist_to_center << " meters away" << std::endl;

            ++frame_count;
        }
    }
};

int main()
{
    for (; restarts_count <= restarts; ++restarts_count) {
        std::cout << "START REALSENSE #" << restarts_count << std::endl;

        try {
            StreamDataController controller;

            controller.init();
            controller.start();

            controller.handle();

            controller.stop();
            controller.release();
        }
        catch (const rs2::error& e) {
            std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
        }
    }
    std::cin.get();

    return 0;
}
```
</details>

_CMakeLists.txt_:
<details> 

```cmake
cmake_minimum_required(VERSION 3.5.0)

set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED TRUE)

set(PROJECT_NAME rs_leak_test)
project(${PROJECT_NAME})

set(CMAKE_CONFIGURATION_TYPES Release)
set(CMAKE_RUNTIME_OUTPUT_DIRECTORY ${CMAKE_BINARY_DIR}/bin)

# Realsense SDK
set(RS_INSTALL_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0")
set(RS_INCLUDE_DIR "${RS_INSTALL_DIR}\\include")
set(RS_BIN_DIR "${RS_INSTALL_DIR}\\bin\\${CMAKE_VS_PLATFORM_TOOLSET_HOST_ARCHITECTURE}")
set(RS_BINS "${RS_BIN_DIR}\\realsense2.dll")
set(RS_LIB_DIR "${RS_INSTALL_DIR}\\lib\\${CMAKE_VS_PLATFORM_TOOLSET_HOST_ARCHITECTURE}")
set(RS_LIBS realsense2)

if(NOT EXISTS ${RS_INSTALL_DIR})
	message("There's no such dir as ${RS_INSTALL_DIR}" FATAL_ERROR)
endif()

include_directories(${RS_INCLUDE_DIR})
link_directories(${RS_LIB_DIR})

add_executable(${PROJECT_NAME} main.cpp)
target_link_libraries(${PROJECT_NAME} ${RS_LIBS})

add_custom_command(TARGET ${PROJECT_NAME}
                   POST_BUILD
                   COMMAND ${CMAKE_COMMAND} -E copy ${RS_BINS} ${CMAKE_RUNTIME_OUTPUT_DIRECTORY}/${CMAKE_CONFIGURATION_TYPES}/)
```
</details>

MartyG-RealSense, 2020-12-30:
Hi @Evlasey   As a starting point, I would recommend removing the **Release()** section that references the instruction **disable_all_streams**.  The Stop() instruction should be sufficient to shut down any active streams on the pipeline and release resources.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline.html#a142ee1adb798f9a03f86eb90895dd8a5

If the pipeline does not close correctly when using Stop() to stop the pipeline then stopping the **sensor** instead of stopping the pipeline may be successful (i.e start the pipeline and stop the sensor).

https://github.com/IntelRealSense/librealsense/issues/7252#issuecomment-695194103

MartyG-RealSense, 2021-01-06:
Hi @Evlasey   Do you require further assistance with this case, please?   Thanks!

Evlasey, 2021-01-06:
Hi @MartyG-RealSense 

Sorry for the late reply. Origginaly I didn't use **disable_all_streams** function in release step (it was empty). I added it to show that even with this function it produce some memoty leaks

MartyG-RealSense, 2021-01-06:
Hi @Evlasey  I had another careful look at your program script.  Normally with librealsense scripts, the **config** statements (also commonly written as cfg) that set up configuration details such as the streams are placed before a pipeline.start() statement.  This means that the configuration is only called **once** when the progra is launched.

I wonder if placing the configuration in the **init()** void is causing the configuration instructions to be called each time that you restart as the program activates each void in the list sequentially.

[image]

Evlasey, 2021-01-08:
Hi @MartyG-RealSense

I need some time to make extra tests. I have no sensors on my hands right now. So I'm going to do this on the next week.

MartyG-RealSense, 2021-01-08:
Thanks very much for the update.   I will keep this case open for a further time period.   Good luck!

Evlasey, 2021-01-11:
Hi @MartyG-RealSense

I prepared some test results with different changes:

 - Original. The sample which I've attached to the issue description. 
   <details>

    [image]
   </details>
   <details>

    ```cpp
    #include <librealsense2/rs.hpp>
    #include <iostream>
    
    static const int restarts = 200;
    static int restarts_count = 0;
    static const int frames_to_restart = 5;
    
    // Global RS variables
    rs2::context g_context;
    std::shared_ptr<rs2::pipeline> g_pipeline = std::make_shared<rs2::pipeline>();
    std::shared_ptr<rs2::config> g_config = std::make_shared<rs2::config>();
    
    class StreamDataController {
    
    public:
    
        void init() {
    
            auto devices = g_context.query_devices();
            
            if (devices.size() == 0)
                throw "Realsense2DepthProvider: Can't enumerate Realsense2 camera";
    
            g_config->enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
        }
    
        void release() {
            g_config->disable_all_streams();
        }
    
        void start() {
            g_pipeline->start(*g_config);
        }
    
        void stop() {
            g_pipeline->stop();
        }
    
        void handle() {
            int frame_count = 0;
            while (frame_count < frames_to_restart) {
                auto frames = g_pipeline->wait_for_frames();
    
                auto depth_frame = frames.get_depth_frame();
    
                // Get the depth frame's dimensions
                float width = depth_frame.get_width();
                float height = depth_frame.get_height();
    
                // Query the distance from the camera to the object in the center of the image
                float dist_to_center = depth_frame.get_distance(width / 2, height / 2);
    
                std::cout << "<Depth sensor> frame=" << frame_count << "\tThe camera is facing an object " << dist_to_center << " meters away" << std::endl;
    
                ++frame_count;
            }
        }
    };
[... 287 lines cut ...]
        g_config->enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
        for (; restarts_count <= restarts; ++restarts_count) {
            std::cout << "START REALSENSE #" << restarts_count << std::endl;
    
            try {
                StreamDataController controller;
    
                controller.init();
                controller.start();
    
                //controller.handle();
    
                controller.stop();
                controller.release();
            }
            catch (const rs2::error& e) {
                std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
            }
        }
        std::cin.get();
    
        return 0;
    }
    ```
   </details>

Evlasey, 2021-01-11:
Found out the source of problems. It's `g_context.query_devices();` function. After commented out `auto devices = g_context.query_devices();` I start getting way better results:
<details>

[image]

</details>

<details>

```cpp
#include <librealsense2/rs.hpp>
#include <iostream>

static const int restarts = 200;
static int restarts_count = 0;
static const int frames_to_restart = 5;

// Global RS variables
rs2::context g_context;
std::shared_ptr<rs2::pipeline> g_pipeline = std::make_shared<rs2::pipeline>();
std::shared_ptr<rs2::config> g_config = std::make_shared<rs2::config>();

class StreamDataController {

public:

    void init() {

        //auto devices = g_context.query_devices();
        
        //if (devices.size() == 0)
        //    throw "Realsense2DepthProvider: Can't enumerate Realsense2 camera";

    }

    void release() {
        //g_config->disable_all_streams();
    }

    void start() {
        g_pipeline->start(*g_config);
    }

    void stop() {
        g_pipeline->stop();
    }

    void handle() {
        int frame_count = 0;
        while (frame_count < frames_to_restart) {
            auto frames = g_pipeline->wait_for_frames();

            auto depth_frame = frames.get_depth_frame();

            // Get the depth frame's dimensions
            float width = depth_frame.get_width();
            float height = depth_frame.get_height();

            // Query the distance from the camera to the object in the center of the image
            float dist_to_center = depth_frame.get_distance(width / 2, height / 2);

            std::cout << "<Depth sensor> frame=" << frame_count << "\tThe camera is facing an object " << dist_to_center << " meters away" << std::endl;

            ++frame_count;
        }
    }
};

int main()
{
    g_config->enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
    for (; restarts_count <= restarts; ++restarts_count) {
        std::cout << "START REALSENSE #" << restarts_count << std::endl;

        try {
            StreamDataController controller;

            controller.init();
            controller.start();

            controller.handle();

            controller.stop();
            controller.release();
        }
        catch (const rs2::error& e) {
            std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
        }
    }
    std::cin.get();

    return 0;
}
```
</details>

So, now I have a question. What should I do to get actual device list without memory leaks?

ev-mp, 2021-01-11:
@Evlasey , the memory leak can be related to the global handle `g_config` that is kept between iterations and that prevents the memory release as it captures device handle internally.
Try to encapsulate `g_config ` and reset it along with `pipe` object for each start.

Evlasey, 2021-01-11:
Extra tests:
 - Reset pipline object
   <details>
   
   [image]

   </details>
   <details>

    ```cpp
    #include <librealsense2/rs.hpp>
    #include <iostream>
    
    static const int restarts = 200;
    static int restarts_count = 0;
    static const int frames_to_restart = 5;
    
    // Global RS variables
    rs2::context g_context;
    std::shared_ptr<rs2::config> g_config = std::make_shared<rs2::config>();
    
    class StreamDataController {
    private:
        std::shared_ptr<rs2::pipeline> _pipeline = std::make_shared<rs2::pipeline>();
    
    public:
    
        void init() {
    
            //auto devices = g_context.query_devices();
            
            //if (devices.size() == 0)
            //    throw "Realsense2DepthProvider: Can't enumerate Realsense2 camera";

        }
    
        void release() {
            //g_config->disable_all_streams();
        }
    
        void start() {
            _pipeline->start(*g_config);
        }
    
        void stop() {
            _pipeline->stop();
        }
    
        void handle() {
            int frame_count = 0;
            while (frame_count < frames_to_restart) {
                auto frames = _pipeline->wait_for_frames();
    
                auto depth_frame = frames.get_depth_frame();
    
                // Get the depth frame's dimensions
                float width = depth_frame.get_width();
                float height = depth_frame.get_height();
    
                // Query the distance from the camera to the object in the center of the image
                float dist_to_center = depth_frame.get_distance(width / 2, height / 2);
    
                std::cout << "<Depth sensor> frame=" << frame_count << "\tThe camera is facing an object " << dist_to_center << " meters away" << std::endl;
    
                ++frame_count;
            }
        }
    };
    
    int main()
[... 295 lines cut ...]
    {
        for (; restarts_count <= restarts; ++restarts_count) {
            std::cout << "START REALSENSE #" << restarts_count << std::endl;
    
            try {
                StreamDataController controller;
    
                controller.init();
                controller.start();
    
                controller.handle();
    
                controller.stop();
                controller.release();
            }
            catch (const rs2::error& e) {
                std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
            }
        }
        std::cin.get();
    
        return 0;
    }
    ```
   </details>

orikama, 2021-01-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 / D435i / D455| 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            | v2.38.1.2223                         | 
| Language                            |  C++                           | 

I've also tested [this](https://github.com/IntelRealSense/librealsense/issues/8091#issue-776493453) sample, on `D435/D435i/D455` cameras, 100 restarts:

<details><summary>D435</summary>

 [image]
</details>
<details><summary>D435i</summary>

  [image]
</details>
<details><summary>D455</summary>

  [image]
</details>

Visual Studio 2019, diff between two memory heap snapshots ( 20 restarts ):
<details><summary>D435</summary>

[image]
</details>
<details><summary>D455</summary>

[image]
</details>

Applying fix similar [to this pull request](https://github.com/IntelRealSense/librealsense/pull/6499) helped with the `std::_Ref_count_obj<librealsense::ptr_option<int>>` leak.

<details><summary>Allocation stack</summary>

[image]
</details>
<details><summary>diff between two memory heap snapshots after fix</summary>

[image]
</details>

But now I don't know what to do with this leak:
<details><summary>Allocation stack</summary>

[image]
</details>

MartyG-RealSense, 2021-01-14:
Hi @orikama  There was a past case involving a memory leak where the backtrace also highlighted the **librealsense::processing_block::invoke** instruction that you highlighted.  Advice was provided on restructing how frames are released.

Evlasey, 2021-01-21:
Hi @MartyG-RealSense 

Sorry for late reply. We shifted focus to other issues. However, we still need further assistance on this issue. We also forgot to mention, that highlighted string were made accidentally (https://github.com/IntelRealSense/librealsense/issues/8091#issuecomment-759975908)

About releasing frames: 
 - in our case we do not use Java wrapper
 - we have no problems with leakage when receiving frames for a long period of time
 - even if we don't receive frames, memory still leaks (https://github.com/IntelRealSense/librealsense/issues/8091#issuecomment-757669205 the last test)

MartyG-RealSense, 2021-01-21:
Are you creating any processing blocks within a streaming loop?

https://github.com/IntelRealSense/librealsense/issues/2115#issuecomment-406873087

Evlasey, 2021-01-25:
@MartyG-RealSense  
> Are you creating any processing blocks within a streaming loop?

No, we don't. We attached minimum source code for every single test we made. So, you can check it by yourself.

Also, the issue which you are linking for is about processing stage. We do not have this kind of problem. Our leak is connected to restarting stream / getting new device list: 
 - https://github.com/IntelRealSense/librealsense/issues/8091#issuecomment-757680316
 - https://github.com/IntelRealSense/librealsense/issues/8091#issuecomment-759975908

MartyG-RealSense, 2021-01-25:
Earlier in this discussion, RealSense team member @ev-mp advised dealing with the memory leak by encapsulating **g_config** and resetting it along with the pipe object for each start.  Were you able to test this advice in your program, please?

https://github.com/IntelRealSense/librealsense/issues/8091#issuecomment-757804485

Hejsil, 2021-01-27:
Not sure this is the same issue but I'm hitting a memory leak using the C API on a simple program like this:
```
#include "librealsense2/rs.h"
#include <stdbool.h>
#include <stddef.h>

int main() {
    while (true) {
        rs2_context *context = rs2_create_context(RS2_API_VERSION, NULL);
        rs2_device_list *devices = rs2_query_devices(context, NULL);
        int count = rs2_get_device_count(devices, NULL);
        for (int i = 0; i < count; i++) {
          rs2_device *device = rs2_create_device(devices, i, NULL);
          rs2_delete_device(device);
        }

        rs2_delete_device_list(devices);
        rs2_delete_context(context);
    }
}
```

To be more clear, it's more like a "thread leak" in that threads are spawned by never torn down causing an explosion of threads until the system runs out of memory. Seems to be the call to `rs2_create_device` causing this.

Evlasey, 2021-01-27:
@MartyG-RealSense 

>Is there any way to restructure your project to avoid the need for restarts?

One of our important clients uses reinitialization of realsense driver routinely during software operation to change settings or device. We can't tell how often it happens.  
Unfortunately, proposed workaround is difficult to implement in our current software as it breaks current control flow.

We are wondering why this simple usage pattern leads to the memory leaks - is it a known issue or it works as expected?

MartyG-RealSense, 2021-01-27:
@Evlasey   Whilst there have been situations where there are problems involving start-stop under certain circumstances - for example, https://github.com/IntelRealSense/librealsense/issues/5713 - I have not identified a case that reminds me precisely of the conditions that your case is experiencing.

There have been cases with problems involving pipeline start-stop that have been addressed by starting the pipeline but stopping the **sensor** instead of stopping the pipeline - i.e start pipeline, stop sensor.

https://github.com/IntelRealSense/librealsense/issues/6459
https://github.com/IntelRealSense/librealsense/issues/6459#issuecomment-648795212

Evlasey, 2021-01-29:
@MartyG-RealSense 
We think our problem is pretty similar to @Hejsil's https://github.com/IntelRealSense/librealsense/issues/8091#issuecomment-768228566. Our test: https://github.com/IntelRealSense/librealsense/issues/8091#issuecomment-757680316

> There have been cases with problems involving pipeline start-stop that have been addressed by starting the pipeline but stopping the sensor instead of stopping the pipeline - i.e start pipeline, stop sensor.

We've tried this possible solution, but it doesn't work.

MartyG-RealSense, 2021-01-29:
Possible problems with query_devices() has also now been reported in https://github.com/IntelRealSense/librealsense/issues/8272

https://github.com/IntelRealSense/librealsense/issues/8272#issuecomment-769812964

Evlasey, 2021-02-08:
Hi @MartyG-RealSense

> Hi @Evlasey Do you require further assistance with this case, please? Thanks!

Yes we do. The problem is still there.

> Possible problems with query_devices() has also now been reported in #8272

In this issue they mentioned (https://github.com/IntelRealSense/librealsense/issues/8272#issuecomment-772595358): 

> Thank you. I see in that case the issue is a memory leak, while what we observed is a memory error.

Also, our issue aren't connected to C# wrapper

Could you suggest any other solution that might solve the problem for our use case?

MartyG-RealSense, 2021-02-14:
Hi @Evlasey   I apologize for the delay in further comment.  I have been seeking advice from Intel about your case.  I do not have an update available yet.

MartyG-RealSense, 2021-02-20:
Hi again @Evlasey   I do not have any new information to provide about your case unfortunately.  Is there any change to your problem if your application is used in the recent **2.42.0** SDK please?

Evlasey, 2021-02-26:
Hi @MartyG-RealSense. We've just tried the lates **Realsense 2.42.0 SDK** with `5_12_7_100` and `5_12_11_0` firmware versions. The memory is still leaking.

MartyG-RealSense, 2021-02-26:
I have highlighted the case to Intel again.  This is the only course of action I can take, as the problem is outside of my own programming knowledge unfortunately.  I do apologize.

Evlasey, 2021-02-26:
@MartyG-RealSense Thanks. We appreciate your assistance.

RealSenseSupport, 2021-03-01:
Hi @Evlasey, thank you for sharing your work and results. Just to double check, it seems the mem leak issue is related to the device context (rs2::context) creation. Or do you also experience the leak with only pipeline start/stop/restart? Thank you.

Evlasey, 2021-03-05:
Hi @RealSenseSupport. Sorry for the delay with this answer.

> Just to double check, it seems the mem leak issue is related to the device context (rs2 :: context) creation.

Yes, the biggest problem at the moment is mem leak on getting device list 

> Or do you also experience the leak with only pipeline start / stop / restart?

We can't say it right now. We'll check everything again and provide our results on the next week

MartyG-RealSense, 2021-03-18:
Hi @Evlasey  Do you have an update that you can provide for @RealSenseSupport about your case, please?  Thanks!

MartyG-RealSense, 2021-03-24:
Hi @Evlasey   Please provide an update for @RealSenseSupport about your case.  The case will be closed in 7 days from the time of writing this if we do not hear from you.  Thank you very much.

VadimAspirin, 2021-03-25:
Hi @RealSenseSupport, sorry for a delayed reply. @Evlasey and I work in one team and now I will answer your question.  
Starting and stopping pipeline with and without catching frames seems to not leak any memory:  
  
<details>

[image]

</details>  
  
<details>

```cpp
#include <librealsense2/rs.hpp>
#include <iostream>

static const int restarts = 200;
static int restarts_count = 0;
static const int frames_to_restart = 5;

// Global RS variables
rs2::context g_context;
std::shared_ptr<rs2::pipeline> g_pipeline = std::make_shared<rs2::pipeline>();
std::shared_ptr<rs2::config> g_config = std::make_shared<rs2::config>();

class StreamDataController {

public:

    void init() {

        auto devices = g_context.query_devices();
        
        if (devices.size() == 0)
            throw "Realsense2DepthProvider: Can't enumerate Realsense2 camera";

        g_config->enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
    }

    void release() {
        g_config->disable_all_streams();
    }

    void start() {
        g_pipeline->start(*g_config);
    }

    void stop() {
        g_pipeline->stop();
    }

    void handle() {
        int frame_count = 0;
        while (frame_count < frames_to_restart) {
            auto frames = g_pipeline->wait_for_frames();

            auto depth_frame = frames.get_depth_frame();

            // Get the depth frame's dimensions
            float width = depth_frame.get_width();
            float height = depth_frame.get_height();

            // Query the distance from the camera to the object in the center of the image
            float dist_to_center = depth_frame.get_distance(width / 2, height / 2);

            std::cout << "<Depth sensor> frame=" << frame_count << "\tThe camera is facing an object " << dist_to_center << " meters away" << std::endl;

            ++frame_count;
        }
    }
};

int main()
{
    StreamDataController controller;
    controller.init();

    for (; restarts_count <= restarts; ++restarts_count) {
        std::cout << "START REALSENSE #" << restarts_count << std::endl;

        try {
            controller.start();

            controller.handle();

            controller.stop();
        }
        catch (const rs2::error& e) {
            std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
        }
    }
    std::cin.get();

    return 0;
}
```

</details>
  
So, the only problem is in device list.

MartyG-RealSense, 2021-04-06:
Hi @Evlasey  @VadimAspirin  Do you still have the memory leak and need advice from @RealSenseSupport please?

MartyG-RealSense, 2021-04-18:
This case will be closed in 7 days from the time of writing this if there are no further comments received.  Thank you.

DixelU, 2021-04-19:
Hi @MartyG-RealSense, sorry for another delay. I work in a team with @Evlasey and @VadimAspirin. I'll answer your questions.  
As of now, the leak is still present and we can't analogously move function `query_devices` out of the loop.

MartyG-RealSense, 2021-04-19:
Hi @DixelU   I will update @RealSenseSupport on the current status of your case - that your remaining problem is believed to be in the device list.

https://github.com/IntelRealSense/librealsense/issues/8091#issuecomment-806969098

yeseuleee, 2021-05-04:
@Evlasey 
How about use static type for your `query_devices()` object.
I have similar problem and I solved this problem like this.
`static devices = g_context.query_devices();`
If you need to call _query_divces()_ repeatedly, it can be an alternative.

RealSenseSupport, 2021-05-11:
@yeseuleee thank you very much for the help to your fix for this case.

DixelU, 2021-05-13:
> How about use static type for your `query_devices()` object.

Sorry, but isn't then `query_devices()` will get called only at the first time of running this code? In case of switching devices, while stream is down, at stream restart - list will not get updated.

RealSenseSupport, 2021-06-04:
A bug has been filed to track this issue internally. It appears the issue is also related to the camera model. The mem leak happens on D455 and D435i but not on D435 or D415. Thank you.

RealSenseSupport, 2021-06-30:
The issue should be addressed in the latest SDK release 2.48.0. 
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2480
Please take a look. Thank you.

closed 2021-07-01 by RealSenseSupport