realsenseai/librealsense #8093: run multiple realsense camera in one arm

Rachel1111, 2020-12-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300                                                    | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   liunx                                                      | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | ARM                                                       |
| SDK Version                            |  2.0                                                        | 
| Language                            |  c++                                                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I need to run three realsense cameras in one ARM, I have tested that if the size of BGR color frame is 1280*720,640*480 or less, it can success. However, when I set the size 1920*1080, the process cannot run. 
Have someone encountered this problem?

MartyG-RealSense, 2020-12-31:
Hi @Rachel1111  If you can successfully run RGB streams on three SR300 cameras at 1280x720 but not at 1920x1080 then that makes me think that it may be a **power draw** issue - that at this maximum RGB resolution, the three SR300s collectively draw too much power for your Arm device to be able to cope with.  

Could you answer the questions below for me please?

-  Are you able to run at 1920x1080 with just one or two cameras instead of three?

-  Are your cameras plugged directly into the USB port of the Arm device, or are they attached to a USB hub?

-  If they are attached to a USB hub, is it a mains electricity powered one?

Rachel1111, 2020-12-31:
Thanks for your reply!
Yean, I can run at 1920x1080 with just one camera, and actually they attached to a USB hub, which is not a mains electricity powered one.

MartyG-RealSense, 2020-12-31:
A USB hub that is not mains powered draws its power from the computer / computing device that it is attached to, in the same way that cameras attached directly to the computer's USB port do.  This can cause problems if the amount of power draw is too high for the computer to cope with.  A mains electricity powered hub provides stable power that is independent of the computer's power supply.

Rachel1111, 2021-01-04:
Thanks for your reply, I can run three cameras when they attached to computer directly. But when they attached to a USB hub, which is a mains electricity powered one, cameras cannot run.

MartyG-RealSense, 2021-01-04:
Are the cameras able to run on a mains powered hub if you boot the computer up without the cameras plugged into the hub, and then insert them int the hub after boot-up has completed, please?

Rachel1111, 2021-01-05:
I've tried this and it still doesn't work, and I can't run three 1280 by 720 cameras right now....

MartyG-RealSense, 2021-01-05:
If you have four USB ports available, you could plug in 3 cameras directly to the computer and put all your other USB devices on the mains powered hub in the fourth computer USB port.  

Or if you only have 3 ports, try 2 cameras directly plugged in, the hub in the third port, and plug the third camera into the hub.

If you are using a computing board based on Arm rather than a full desktop / laptop computer then its resources may be being over-burdened by three cameras though if all three are active at the same time.   An exception to that would be Arm-based Nvidia Jetson boards such as TX2 and Xavier, which should be better able to handle the demands of simultaneously active multiple cameras.

Another complicating factor with having all SR300 cameras active would be that multiple SR300s interfere with each other if they are positioned close enough for their projections to overlap.  The more modern RealSense 400 Series cameras do not interfere with each other.

Rachel1111, 2021-01-06:
Ok, Thanks for your answers. I'll try your suggestions.

Rachel1111, 2021-01-06:
I'm sorry to ask you another question, why can't I even run three 640 x 480 cameras now? Is there a need to reset the realsense camera or something?

MartyG-RealSense, 2021-01-06:
Are the three SR300 cameras being detected at all?

Rachel1111, 2021-01-06:
Yeah, they are being detected. But they cannot be initial, and report mistakes "Frame didn't arrive within 15000".

MartyG-RealSense, 2021-01-06:
SR300s may sometimes have problems with disconnecting immediately or very soon after activation of a stream, even though they can be detected.  This may be due to issues related to the stability of power being supplied to the cameras from USB.  I would normally suggest using a mains powered USB hub instead of plugging the cameras directly into the computer, but you have already tried that and still had problems.

A way to test whether it is a power-related problem would be to set their **Laser Power** setting in the RealSense Viewer to zero whilst the depth stream is off and see whether the camera remains connected when the depth stream is activated.   Do not be concerned if you do not see a depth image, as this is normal when Laser Power is minimized.

[image]

If the camera remains active after stream activation when Laser Power is zero, left-click anywhere on the Laser Power slider to increase its value.   If the camera then disconnects, this will suggest that there is an issue with power supply to the camera (that it is not receiving enough to meets its current power needs and maintain the connection).

Rachel1111, 2021-01-07:
I have tried Arm-based Nvidia Jetson boards such as TX2 and Xavier, it still cannot work. And I am using two Arm-based boards now.

Rachel1111, 2021-01-07:
Thanks very much!

MartyG-RealSense, 2021-01-07:
A turning point during this case seems to be when you were able to successfully run three SR300 cameras, and then later weren't able to.  So something clearly changed.  If all three cameras suddenly can not activate, this would suggest that the problem is not related to the cameras.

When you add the camera to the RealSense Viewer,  can you confirm please that you are only activating the **RGB Camera** module?  Activating the **Stereo Depth** module is likely to be the trigger for a disconnection of the camera if its power needs are not being met.

In the image below, I stream 1920x1080 RGB from an original SR300 (recognised as SR305) and plugged into a mains powered USB hub.

[image]

closed 2021-01-20 by MartyG-RealSense