realsenseai/librealsense #8094: Is there a simple way to get pose data from D435I's gyro and accel?

Skylark0924, 2020-12-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |    Win 10  |                               | 
| Platform                                 | PC  |
| SDK Version                            |   2. }                          | 
| Language                            |  python                          | 
| Segment			|  Robot                           | 

### Issue Description
Is there a simple way to get pose data from D435I's gyro and accel?

MartyG-RealSense, 2020-12-31:
Hi @Skylark0924   The D435i does not have built-in support for 6 degrees of freedom (6-DOF) pose data.  The RealSense SDK sample program **rs-motion** does though provide calculation of the rotation of RealSense camera models that are equipped with an IMU (D435i and D455) 

https://github.com/IntelRealSense/librealsense/tree/master/examples/motion

Skylark0924, 2021-01-04:
Thanks a lot for your explicit explanation!
I got a very accurate rotation via **rs-motion.exe**, and I will find the python-based solution later.

However, I wonder if I can get a relative position about the camera.
For example, I set the initial 3D position as (0, 0, 0), and get the relative position after motion. 
Is this possible? Or I can just use the time integration of the accelerometer.

MartyG-RealSense, 2021-01-04:
The link below contains a guide based around ntel's D435i SLAM tutorial for ROS.

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM

In the section of the guide headed **Localization with rtabmap**, it suggests using the software **RTABMAP** to study a recorded pointcloud map file and obtain the relatove position from that map.

Skylark0924, 2021-01-05:
Thanks a lot!

closed 2021-01-05 by Skylark0924