realsenseai/librealsense #8095: Real-time tracking of facial marker points

nevynwong, 2020-12-31:
| Camera Model                       | D415| 
| Operating System & Version |   win10  | 
| Platform                                 | PC |    | 

The SDK includes a program for face tracking, and I'm wondering if the facial marker points are just 2D? Are they captured in real time? I want to use the camera to accomplish the function of capturing marker points in real time, but I don't know how to do it? Can you help me?

MartyG-RealSense, 2020-12-31:
hI @nevynwong   If you are referring to the SDK's **dlib** facial landmark detection sample, its documentation states that the landmarks are 68-point 2D coordinates.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/dlib/face

The dlib library usually operates in real-time, as do other facial landmark detection systems.

Information about setting up the RealSense dlib wrapper can be found in the link below:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/dlib

If you are using Python, information about facial landmark detection with Python can be found here:

https://support.intelrealsense.com/hc/en-us/community/posts/360038289934/

nevynwong, 2021-01-02:
Thanks, is it not possible to get 3D for the 68 points captured? Or can I acquire both 2D and 3D images and then map the 2D points to 3D.

MartyG-RealSense, 2021-01-02:
The link below is a project that demonstrates using dlib for 3D facial landmark detection:

https://github.com/ostadabbas/3d-facial-landmark-detection-and-tracking

nevynwong, 2021-01-03:
Does that mean that such 3D points are not obtained in real time?

MartyG-RealSense, 2021-01-03:
Although the 2D landmark detection is undoubtedly real-time, is is less obvious when a 2D to 3D landmark project's 3D reconstruction from the landmarks is in real-time.

Of the projects that did describe their functions clearly enough to reach a conclusion about how they render 3D, the majority seemed to apply 3D landmarks to pre-recorded files such as 3D mesh models of faces and ply point cloud files, like the Python examples in the links below:

https://github.com/LightOfMonet/3Dface_landmark

https://github.com/RasmusRPaulsen/Deep-MVLM

Of the projects that clearly described having real-time 3D reconstruction for the face, **OpenPose** was the best solution that my research could find.

https://github.com/CMU-Perceptual-Computing-Lab/openpose

[image]

The Python project **3DDFA_V2** also demonstrates building up a 3D face mask from detected 2D landmarks points.

https://github.com/cleardusk/3DDFA_V2

nevynwong, 2021-01-04:
Hi, @MartyG-RealSense, Is it normal for the depth map to keep flickering in the captured video stream?

MartyG-RealSense, 2021-01-04:
Hi @nevynwong   Do you have fluorescent lights such as ceilng strip lights present in the scene where you are using the camera, please?  These lights contain a gas that flickers at a frequency that is difficult to see with the human eye and can cause noise on a camera image.

If you do have such lights, using an FPS speed that is close to the operating frequency of the lights may help to negate the disruption.  For some lights the appropriate FPS speed may be 30 FPS and for other lights it may be 60 FPS.

closed 2021-01-17 by MartyG-RealSense