realsenseai/librealsense #8096: undefined symbol: libusb_get_port_numbers

sw14928, 2020-12-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 19) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Trying to run my code (which previously worked), my code now stops on the import pyrealsense2 line, giving me the following error:

    import pyrealsense2 as rs
  File "/usr/local/lib/python2.7/dist-packages/pyrealsense2/__init__.py", line 2, in <module>
    from .pyrealsense2 import *
ImportError: /usr/local/lib/python2.7/dist-packages/pyrealsense2/pyrealsense2.so: undefined symbol: libusb_get_port_numbers

I have no idea how to fix this. I have tried uninstalling and reinstalling. Can anyone help please?

RealSenseSupport, 2021-01-13:
Hi @sw14928 

Were you able to get past your error?  Were you able to setup the environment based on the instructions:  https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#python-wrapper

sw14928, 2021-01-13:
Hi, Unfortunately not. I have been trying for days now. Can anyone help, please?

RealSenseSupport, 2021-01-20:
Hi @sw14928 

When you say "which previously worked" could you provide some details on what changed?

Thanks

closed 2021-03-24 by RealSenseSupport