realsenseai/librealsense #8098: Hunk #1 FAILED at 153. 1 out of 1 hunk FAILED -- saving rejects to file drivers/media/usb/uvc/uvcvideo.h.rej

jindanChina, 2021-01-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i |  
| Operating System & Version |Ubuntu 16.04| 
| Kernel Version (Linux Only)    |  4.15.0-128-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.41.0 | 

### Issue Description
<初始化空的 Git 仓库于 /home/jindan/librealsense/ubuntu-xenial-hwe/.git/
Fetching Ubuntu LTS tag Ubuntu-hwe-4.15.0-128.131_16.04.1  to the local kernel sources folder
remote: Counting objects: 5930029, done.
remote: Compressing objects: 100% (867979/867979), done.
remote: Total 5930029 (delta 5025768), reused 5919832 (delta 5018658)
接收对象中: 100% (5930029/5930029), 1.26 GiB | 386.00 KiB/s, 完成.
处理 delta 中: 100% (5025768/5025768), 完成.
来自 git://kernel.ubuntu.com/ubuntu/ubuntu-xenial
 * [新标签]          Ubuntu-hwe-4.15.0-128.131_16.04.1 -> Ubuntu-hwe-4.15.0-128.131_16.04.1
Switching to LTS tag Ubuntu-hwe-4.15.0-128.131_16.04.1
正在检出文件: 100% (63727/63727), 完成.
Note: checking out 'Ubuntu-hwe-4.15.0-128.131_16.04.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD 目前位于 2c5bb3a... UBUNTU: Ubuntu-hwe-4.15.0-128.131~16.04.1
Applying patches for xenial-hwe line
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 208 with fuzz 2 (offset 5 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 FAILED at 153.
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/usb/uvc/uvcvideo.h.rej
patching file drivers/media/v4l2-core/v4l2-ioctl.c
Hunk #1 succeeded at 1248 (offset 2 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 680 (offset 12 lines).>
I've been puzzling this problem for a long time. I hope I can get help！

MartyG-RealSense, 2021-01-04:
Hi @jindanChina   Another RealSense user experienced this same set of errors and offered advice for how to correct it in the link below.

https://github.com/IntelRealSense/librealsense/issues/7863#issuecomment-736776850

jindanChina, 2021-01-05:
@MartyG-RealSense Thank you very much, the above problem has been solved very well. But I encountered some other problems, I am trying to update the firmware of 435i to see if it can be solved, thank you again for your answers.

jindanChina, 2021-01-05:
@MartyG-RealSense After updating the firmware, many problems were solved, but another problem still exists. Can you help me solve it?
05/01 11:11:44,943 WARNING [140610460559104] (sensor.cpp:1086) RGB Camera: skipping Backlight Compensation control. descriptor: [min/max/step/default]= [0/0/0/0]
 05/01 11:11:44,943 WARNING [140610460559104] (sensor.cpp:1086) RGB Camera: skipping Exposure control. descriptor: [min/max/step/default]= [0/0/0/0]
 05/01 11:11:44,944 ERROR [140610460559104] (types.h:313) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 05/01 11:11:44,944 WARNING [140610460559104] (sensor.cpp:1110) Failed to add Enable Auto Exposure control for RGB Camera
 05/01 11:11:55,198 WARNING [140610635167552] (ds5-motion.cpp:477) IMU Calibration is not available, default intrinsic and extrinsic will be used.

MartyG-RealSense, 2021-01-05:
Hi @jindanChina   I suspect from those new error messages that there are still problems with the kernel patching.  A way to confirm this would be to build libralsense with CMake using the **RSUSB** installation method.  It requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.

Instructions for the RSUSB installation method can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-665650908

I would recommend leaving out the term **-DBUILD_WITH_CUDA:bool=true** from the CMake build instruction in the above link, as you may not have an Nvidia GPU graphics chip in your computer.

ev-mp, 2021-01-06:
@jindanChina hello,  instead of running the scripts manually you can use DKMS Debian:
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution
```sh
sudo apt install librealsense2-dkms
```

As for the manual patch  - it will be handled and update accordingly.

jindanChina, 2021-01-07:
@ev-mp @MartyG-RealSense Thank you, I will try your suggestions, you are really enthusiastic, thank you!

closed 2021-01-20 by MartyG-RealSense