realsenseai/librealsense #8100: WARNING [547852988432] (ds5-factory.cpp:1123) DS5 group_devices is empty.

punchnami, 2021-01-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (05.12.09) | 
| Operating System & Version |   Linux (Ubuntu 18.04.5)   | 
| Kernel Version (Linux Only)    |  (4.9.14)                                         | 
| Platform                                 | NVIDIA Jetson NANO  |
| SDK Version                            |  2.0                          | 
| Segment			|  {Robot}                          | 

### Issue Description
<My Intel realsense D435i doesn't work with Jetson NANO but it can work with my laptop(window) just fine>

Note: - USB type description: 3.2
          -I already install librealsense
          
Here's the log when i ran the command realsense-viewer

`jsnn@jsnn:~/librealsense$ realsense-viewer
 04/01 14:29:51,540 WARNING [547852988432] (ds5-factory.cpp:1123) DS5 group_devices is empty.
 04/01 14:29:52,178 WARNING [547852988432] (ds5-factory.cpp:1123) DS5 group_devices is empty.
 04/01 14:29:52,396 WARNING [547852988432] (ds5-factory.cpp:1123) DS5 group_devices is empty.
 04/01 14:29:52,613 WARNING [547852988432] (ds5-factory.cpp:1123) DS5 group_devices is empty.
 04/01 14:29:52,828 WARNING [547852988432] (ds5-factory.cpp:1123) DS5 group_devices is empty.
 04/01 14:29:53,046 WARNING [547852988432] (ds5-factory.cpp:1123) DS5 group_devices is empty.
 04/01 14:29:53,263 WARNING [547852988432] (ds5-factory.cpp:1123) DS5 group_devices is empty.
 04/01 14:29:53,479 WARNING [547852988432] (ds5-factory.cpp:1123) DS5 group_devices is empty.
`
Any help would be appreciated!

P.S. I want to apply D435i for my robot's SLAM

MartyG-RealSense, 2021-01-04:
Hi @punchnami   This is an issue that is probably related to your librealsense installation.   I would recommend starting with a fresh installation of librealsense o your Jetson.

One RealSense user who had this particuar error solved it by using the Nano guide on the **JetsonHacks** website:

https://www.jetsonhacks.com/2019/12/22/install-realsense-camera-in-5-minutes-jetson-nano/

Another installation guide that has also worked consistently for Nano users is the one detailed in the link below for building librealsense and the Python wrapper together from source code with the **RSUSB** method:

https://github.com/IntelRealSense/librealsense/issues/7905#issuecomment-737544294

punchnami, 2021-01-05:
Thank you for your help @MartyG-RealSense !

The first one using the Nano guide still giving the same error even though I freshly installed it. I tried the second method and received the following error: 

`jsnn@jsnn:~$ python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense as rs
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pyrealsense'
>>> import pyrealsense2 as rs
>>> pipe = rs.pipeline()
>>> profile = pipe.start()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: No device connected
`

I did connect my D435i with Jetson Nano, is there other ways? This is my usb log

`jsnn@jsnn:~$ lsusb
Bus 002 Device 003: ID 8086:0b3a Intel Corp. 
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 005: ID 2a7a:6a18  
Bus 001 Device 004: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

Thank you in advance :)

MartyG-RealSense, 2021-01-05:
The official Intel guide for RealSense installation on Jetson is in the link below.   It is not specifically for Nano but has been tested with it, and I have tested it successfully on a Nano myself today.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

punchnami, 2021-01-06:
Thank you @MartyG-RealSense , I will check this method out.

punchnami, 2021-01-11:
@MartyG-RealSense 

I'm getting the error `E: Unable to locate package librealsense2-dkms`  while installing the Debian Package with the commadn `sudo apt-get install librealsense2-dkms` according to this installation guide  https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages.

Thank you for your help!

MartyG-RealSense, 2021-01-11:
Hi @punchnami   The package installation method in the above link will not work on devices with an Arm processor chip such as Jetson, as it is meant for x86 /x64 processors like those in desktop and laptop PCs.  Jetson has its own package installation method:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

punchnami, 2021-01-11:
Thank you for your reply @MartyG-RealSense , I was a bit confuse. To install the debian package, I have to follow the Building from Source using RSUSB Backend installation guide from the link you provided, right?

MartyG-RealSense, 2021-01-11:
The two most common methods of installing librealsense are to install from packages **or** build it from source code.  They are two very different methods.  The RSUSB method builds from source code and does not involve packages.

punchnami, 2021-01-11:
Thank you so much for your helps @MartyG-RealSense !!! I deleted all my previous installation and installed librealsense again by build it from source code and I'm able to run realsense-viewer this time! It works!

closed 2021-01-11 by punchnami

MartyG-RealSense, 2021-01-11:
Excellent news that you got it working @punchnami - thanks very much for the update!