realsenseai/librealsense #8101: How to merge depth data for real-time dense and accurate 3D reconstruction

CarMineo, 2021-01-04:
| Required Info                        |                                                                |
|--------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | Firmware 05.12.07.100 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                           | 2.0 | 
| Language                               | C/C#/C++/labview/matlab | 
| Segment			         | Robot | 

### Issue Description
Dear all,

Happy New Year.

I am sorry if I am duplicating an issue thread here. I am trying to develop a dense 3D reconstruction application for a D435i camera manipulated by a robot arm. I am new in this area, so I have been exploring the literature and I have found a plethora of publications. The authors of some scientific publications describe exactly what I am trying to achieve, which is accurate and dense 3D reconstruction of small-medium size objects (e.g. 500mm long at the most). Some works have adapted SLAM to depth cameras, optimizing how depth points from different poses are merged into a single representation, minimising the drift and the stitching errors (e.g. https://www.microsoft.com/en-us/research/wp-content/uploads/2016/02/ismar2011.pdf and https://dl.acm.org/doi/10.1145/3182157).
Despite of being able to read the scientific publications, I have not found a straightforward open source application I can use with the D435i camera yet. Most researchers describe how to estimate the camera poses from the depth data. However, this is not a problem for me, since my D435i camera is not hand-held. It is manipulated by a robot, so I can use the robot feedback positions to rotate and translate the depth data. Instead, I would like to understand how to merge depth data coming from different poses in real-time, in order to incrementally get a dense and accurate 3D reconstruction of the scene.

Hope to hear from you soon.
Thanks in advance,
CarMineo

MartyG-RealSense, 2021-01-04:
Hi @CarMineo   It does sound as though the appropriate scanning solution for your single-camera setup is one where the camera is moved around and the detail of the image is progressively built up until it reaches a state where you are satisifed with it,  There are a number of RealSense-compatible commercial scanning software solutions that can do this.  

https://support.intelrealsense.com/hc/en-us/community/posts/360037943654/comments/360009971393

https://support.intelrealsense.com/hc/en-us/community/posts/360051853333/comments/360014488833

Of the open-source approaches available, a good starting point would be Intel's own D435i SLAM tutorial, which enables you to use ROS to build up a point cloud and save the point cloud data to a **pcd** file.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

If your robot arm requires hand-eye calibration, this link is worth reading:

https://support.intelrealsense.com/hc/en-us/community/posts/360051325334/comments/360013640454

Using a **gimbal** mount with 400 Series cameras for robotics applications to provide a tilt mechanism has also been discussed in the past.

https://support.intelrealsense.com/hc/en-us/community/posts/360052504573-Gimbal-for-Realsense-cameras

MartyG-RealSense, 2021-01-10:
Hi @CarMineo   Do you require further assistance with this case, please?  Thanks!

CarMineo, 2021-01-10:
Hello @MartyG-RealSense,
Thanks for your help and your detailed reply. It has been very useful. I do not need further assistance with this.
Thanks,
CarMineo

closed 2021-01-10 by CarMineo

MartyG-RealSense, 2021-01-10:
Great to hear that you found the information helpful @CarMineo - thanks for the update!