realsenseai/librealsense #8102: realsense-net device in C#

AlexisC34, 2021-01-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515| 
| Firmware Version                   |01.05.02.00| 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| Language                            |  C#| 

### Issue Description
Hello,

Is it possible to connect to a Raspberry running rs-server in C# ? I know we can do it in cpp with this tutorial : https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

With these lines :
1. Add #include <librealsense2-net/rs_net.hpp> to the code
2. Inside CMakeLists.txt add realsense2-net to target_link_libraries
3. Create network device from IP and add it to a context:
         rs2::net_device dev("10.0.0.99");
         rs2::context ctx; dev.add_to(ctx);
4. Pass context to pipeline constructor:
         rs2::pipeline pipe(ctx);

I don't find any C# wrapper that can do it. 

My goal is to use the camera over ethernet (for long distances > 50 meters). Is it possible?
Or do you have any other solution?

Regards,
Alexis

RealSenseSupport, 2021-01-13:
Hi @AlexisC34 

Our solution is based on what is outlined in the whitepaper tutorial.  We haven't experimented outside of that.

RealSenseSupport, 2021-01-29:
Hi @AlexisC34 

Were you able to get your setup working?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-02-05 by RealSenseSupport

AndreV84, 2021-05-23:
how to use realsense-2 net with L515?