realsenseai/librealsense #8104: How to improve long range depth?

zapaishchykova, 2021-01-04:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435/D435i } | 
| Firmware Version                   | (Open RealSense Viewer v.2.38.1) | 
| Operating System & Version |   {Win 10}  | 
| Platform                                 | PC   |
| Language                            |  {python}                          | 

### Issue Description
Hello! I am currently experimenting with capturing the depth of the whole room ( not too big one, 4 by 6 meters). And what i see is the very unpleasant noisy depth after approximatelly 3 meters. 
What i have tried from this document https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf:
- Increasing depth units.
- Increasing laser power to maximum. 
Still, even with postprocessing, the texture of depth is very noisy and "bubbly" on the long-distance. Are there any more suggestions on how to maybe make the depth less precise, but more uniform?
[image] Thanks!

MartyG-RealSense, 2021-01-04:
Hi @zapaishchykova   Firstly, if you are seeking to capture a full 6 meters distance in the RealSense Viewer then you should ensure that a post-procesing filter called **Threshold Filter** is disabled or set to a value of '6' or more.  By default this filter is set to '4' meters and restricts the maximum range of the rendered depth data to that distance if the filter is enabled and left on its default setting.

Scripts written in the SDK, such as in the pyrealsense2 wrapper for Python, do not contain post-processing by default, and such filters must be deliberately programmed into scripts.  So the output generated by a RealSense user's own applications will tend to look different to the output produced by the Viewer (which has a range of  post-processing filters enabled by default).

[image]

It should not be necessary to increase the **Depth Units** scale from its default of '0.001'.

Yes, performing a calibration of the camera may improve results.  For a robust and advanced calibration of the camera, I would recommend using the **Dynamic Calibrator** tool.

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

If you are capturing a low-texture white wall at the far end of the scene, adding a type of physical optical filter called a **long-pass filter** over the top of the camera lenses on the outside of the camera may make distant white walls more visible.  This is described in section **4.2.1  Conditions with no natural texture** of Intels white-paper document about optical filters.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#section-4-the-use-of-optical-filters

zapaishchykova, 2021-01-04:
Hi @MartyG-RealSense,
Thank you for your always fast reply!

1. Yes, depth threshold filters are disabled, and yes, in python wrapper code i checked that it has all the same preprocessing steps ( i am using box-dim. multicam demo with few adjustments https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam) 

2. I saw this whitepaper with long-pass filter, do I understand it correctly that it's some kind of hardware? Does intel cell those or do you have any brand recommendations?

MartyG-RealSense, 2021-01-04:
Yes, optical filters are physical hardware.  They are not supplied by Intel.  The Chief Technical Officer of the RealSense Group at Intel (**agrunnet**) provides some recommendations for filter suppliers at the link below:

https://github.com/IntelRealSense/librealsense/issues/2875#issuecomment-446793681

MartyG-RealSense, 2021-01-10:
Hi @zapaishchykova   Do you require further assistance with this case, please?  Thanks!

zapaishchykova, 2021-01-11:
Hi @MartyG-RealSense ,
So I have tried re-calibration with dynamic callibration (it did not help). The cost of long-pass filters is unfortunately is way out of my budget (400 euros on average per one, and if you consider that you need 2 per each IR emitter that becomes way too expensive). 

I have been reading on how to tune cameras for the best long-range performance and saw a few issues with the same problem: 
- https://github.com/IntelRealSense/librealsense/issues/4553
- https://github.com/IntelRealSense/realsense-ros/issues/1138
- https://github.com/IntelRealSense/librealsense/issues/1375

Seems like there is nothing really you can do to capture something (without super wavy artefacts) that is further than 2m, otherwise the texture becomes really wavy and hard to work with. I would probably try another depth sensors!

closed 2021-01-11 by zapaishchykova