realsenseai/librealsense #8106: export_ply_example.py wrong

Steven-ZY, 2021-01-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455| 
| Firmware Version                   |05.12.09.00) | 
| Operating System & Version |   {Win 10 and Ubuntu18.04  | 
| Platform                                 | PC  |
| Language                            |  python                          | 

### Issue Description
Hey, guys! You know, I run the program 'export_ply_example.py'[https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py) successfully every time on my PC with win10. But on Ubuntu18.04, it was always wrong. I can't find the answer.

[image]
[image]

MartyG-RealSense, 2021-01-04:
Hi @Steven-ZY   There was a past case involving **export_ply_example.py** that produced a similar output on Ubuntu.  The RealSense user in that case was advised that it may be due to depth **outliers** that may be confusing the scale and should be filtered out if possible.

https://github.com/IntelRealSense/librealsense/issues/3579

Steven-ZY, 2021-01-05:
@MartyG-RealSense Thank you for your rapid reply. However, I still can't solve the problem. I'm confused about it. The same script, the different output in Win10 and Ubuntu. Could you give me more advice, please?

MartyG-RealSense, 2021-01-05:
Effects may sometimes appear on Ubuntu that are not encountered when running the same program on Windows, and may be related to conflicts with the Linux kernel on a particular computer.  Building librealsense on Ubuntu with the **RSUSB** installation method often corrects problems related to the kernel, as this method is not dependent on Linux versions or kernel versions and does not require patching.

closed 2021-01-19 by MartyG-RealSense