realsenseai/librealsense #8107: Can d435 camera be controlled by cable?

wuhaoqi-hub, 2021-01-05:
Can d435 camera be controlled by cable?typec-cable

MartyG-RealSense, 2021-01-05:
Hi @wuhaoqi-hub  The D435 uses a Type-C USB cable that is supplied with the camera when it is purchased.

If this is not what you meant, please provide further details about 'controlling' the camera with a cable.   Thanks!

wuhaoqi-hub, 2021-01-05:
Sorry, I didn't ask clearly. What I want to ask is whether the camera can be powered and controlled by network cable?

MartyG-RealSense, 2021-01-05:
The Intel RealSense 400 Series cameras do not have built-in Power over Ethernet (PoE) connectors.  You can purchase PoE 'hats' for boards such as Raspberry Pi to supply and control the power of the board that the camera is attached to.

https://github.com/IntelRealSense/librealsense/issues/8088#issuecomment-752401552

Third-party industrial RealSense 400 Series compatible cameras provided by the companies FRAMOS and LIPS Corp have built-in power over ethernet connectors.  As these cameras are not Intel products, Intel does not provide technical support for them though and support is the responsibility of the particular manufacturer.

wuhaoqi-hub, 2021-01-05:
thank you for your reply!

closed 2021-01-05 by wuhaoqi-hub