realsenseai/librealsense #8110: No 3D point cloud in RealSense Viewer

AlexXieee, 2021-01-05:
| | |
|---|---|
|**librealsense**|2.41.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D455|
|**Serial Number**|035322250100|
|**Firmware Version**|05.12.10.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|025523050865|
|**Firmware Update Id**|025523050865|

Hello, the D455 is connected to a Thinkpad X1 Carbon Gen 8, using windows installer. The 3D cloud point appears only 3-5 seconds after open the Viewer, and then disappear with only axis left; the 2D mode still works in stereo mode and in RGB mode. This problem only appears on my computer. I re-install the software but make no sense.

MartyG-RealSense, 2021-01-05:
Hi @AlexXieee   I can foresee a couple of situations where you may get only the axis in 3D point cloud mode:
 
-  If the Stereo Module and RGB Camera modules are not currently enabled when you switch to 3D mode (the icon beside the modules is colored red, meaning that they are off).

[image]

-  If the RGB Camera module is currently enabled but the Stereo Module is not enabled.

[image]

If you have at least the Stereo Module enabled when in 3D mode then you should receive a point cloud.

[image]

AlexXieee, 2021-01-05:
Hi @MartyG-RealSense , thanks for your fast reply. However, it seems not as simple as you describe. Let me show you the video so that you can understand my means. Also, I found even though there are no 3D cloud point, the RAM consumption will reach up to 10GB after several minutes, which seems like a bug. 

https://user-images.githubusercontent.com/51752769/103651120-c29a8900-4f69-11eb-8823-3e6672348935.mp4

AlexXieee, 2021-01-05:
Hi @MartyG-RealSense , previously I installed the full SDK for WIN10. And now I download only Viewer.exe and open it. The same problem appears: the cloud point flickers and then disappears.

MartyG-RealSense, 2021-01-05:
Can you test with the **rs-pointcloud.exe** example program?   You can find this in the **Tools** folder of the SDK by right-clicking on the launch shortcut for the Viewer and selecting the **Open file location** menu option.    If rs-pointcloud can generate a point cloud without problems for an extended period of time then it may suggest that the problem you are experiencing is with the Viewer program specifically.

[image]

[image]

AlexXieee, 2021-01-05:
Hi @MartyG-RealSense , there is no problem when I use **rs-pointcloud.exe** so that the problem is resulted from Viewer programme.

MartyG-RealSense, 2021-01-05:
Thanks again for the test update.   Could you download the **2.40.0** standalone Viewer version please to test whether or not the problem is with the latest **2.41.0** Viewer?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.40.0

AlexXieee, 2021-01-05:
Hi @MartyG-RealSense, I tried 2.40.0, 2.39.0, 2.38.1, and 2.37.0, all of them appears the same problem.

MartyG-RealSense, 2021-01-05:
I have not encountered an issue like the one that you are experiencing.   There was a recent case though where a RealSense user could not run the Viewer correctly because the **GLSL** graphics functions were enabled in the Viewer's settings.  They did not have similar problems in SDK programs that did not use GLSL.  So let's try disabling GLSL for the sake of eliminating it as a cause of the problem.

1.  Left-click on the gear-wheel icon in the top corner of the Viewer and select the **Settings** option to open the Viewer's settings interface.

[image]

2.  Select the **Petformance** option tab of the settins window and left-click on the blue boxes beside the 'Use GLSL for rendering' and 'Use GLSL for processing' options, so the boxes turn from blue (On) to black (Off).  Then click the **Apply** button at the botom of the settings interface.

[image]

AlexXieee, 2021-01-05:
Hi @MartyG-RealSense, I have completed the test in the Use GLSL for Rendering and Use GLSL for Processing, and the result is:
1. Enable Use GLSL for Rendering, no problem, normal RAM usage;
2. Enable Use GLSL for Processing, no problem, but the RAM usage start to rapidly increase;
3. Disable both options, no problem; normal RAM usage;
4. Enable both options, problem appears (the original issue).
Much appreciation for your guidance, and I wonder what the diffrence is, for using / unusing GLSL. Does GLSL affect generated PLY file or BAG file, that is, the quality or information?

MartyG-RealSense, 2021-01-05:
GLSL is an optional graphics acceleration feature that can "offload" processing work from the CPU to the graphics GPU for some SDK functions such as point clouds.  It may not produce noticable improvement on low-end computers though, and you should still get good performance from the SDK if it is not used.

The link below has a good guide to GLSL.

https://github.com/IntelRealSense/librealsense/pull/3654

closed 2021-01-05 by AlexXieee